mongoose-os-apps/shelly-homekit #1157: roller shutter on 2.5 gets stuck

oliverrahner, 2022-12-16:
I recently outfitted a friends house with 10 Shelly 2.5 PM for roller shutter control.
Two of them are currently showing the following behavior.
They
* are displayed as "closed" in Homekit
* are really "open"
* don't react to the physical buttons

No action is possible anymore, neither via Homekit, nor via the wall buttons.

The only recovery I found so far is to start a new calibration, but for the ones that show this behavior it returns latest after a few days.

I have yet to look at the logs, because I only sporadically have access to the devices, but will add them here once I got them.

jadaka23, 2022-12-16:
Same Issue here with only one of my Shelly 2.5 in Roller shutter Mode. Only Way to solve is to recalibrate

oliverrahner, 2022-12-16:
Maybe the issue is related to the fact that I have set up the device completely on the stock firmware and then migrated to Shelly HomeKit later.
Maybe some settings from the stock firmware are interfering, if they are not removed correctly?

marwonn, 2022-12-17:
> I recently outfitted a friends house with 10 Shelly 2.5 PM for roller shutter control. Two of them are currently showing the following behavior. They
>
> * are displayed as "closed" in Homekit
> * are [...]

I've been experiencing the exactly same behaviour with one out of my five Shelly 2.5. Since yesterday they behave as @oliverrahner describes it:

- are displayed as "closed" in Homekit
- are really "open"
- don't react to the physical buttons

I didn't change anything - appeared out of the blue.

marwonn, 2022-12-17:
> Same Issue here with only one of my Shelly 2.5 in Roller shutter Mode. Only Way to solve is to recalibrate

Doesn't work for me. The recalibration stops at approximately 50 %, so it doesn't fully close. After recalibration the 'Status' displays: 'idle pos 49' - but the roller shutter is fully open.

oliverrahner, 2022-12-21:
I fixed both devices again today by recalibration.
Next, I tried to hard reset both. Turns out: "Reset" doesn't wipe anything, the device just reboots and the whole configuration remains in place...

marmitol, 2023-01-03:
I have the same problem as above.

Even the physical buttons do not work. The solution is to calibrate.

oliverrahner, 2023-01-03:
> I managed to reset the devices by reverting to stock, factory reset, upgrade to homekit firmware again :) Let's see what happens next...

The error is still present, at roughly the same rate as before.
I'm still looking to stuff those Wifi white spots and see if it gets more stable then.

timoschilling, 2023-01-16:
@oliverrahner And other Shelly 2.5 in WC mode didn’t have that problem?

oliverrahner, 2023-01-16:
One has it very regularly, another one not as often, the other 7 not at all.
That's why I attributed it to Wifi coverage in the first place...

timoschilling, 2023-01-16:
How old that devices? All (working and broken) same age? Does all hat a version < 2.10 installed bevor?

oliverrahner, 2023-01-16:
The devices are brand new (< 4 months).
The all started with 2.11.2

oliverrahner, 2023-02-17:
I have "fixed" my problem by moving back to the original firmware and using Homebridge in between 😕

closed 2023-02-17 by timoschilling

marmitol, 2023-02-18:
The problem is not solved. I applied as a solution back to the previous version(Firmware Version:2.11.1). With this version it works correctly. Please check this latest version, as it is not working properly. Thanks.

Andi161, 2023-06-10:
>I applied as a solution back to the previous version(Firmware Version:2.11.1). With this version it works correctly.

Andi161, 2023-06-10:
> I applied as a solution back to the previous version(Firmware Version:2.11.1). 
I reverted all my 2.5 back to 2.11.1, but one particular shelly still has this "stuck" problem. For 4 other Shellies it solved the problem. I am trying now to disable power-save mode on the failing shelly and see what happens.