mongoose-os-apps/shelly-homekit #1158: Edge Switch Mode is not available

SunseekerFfm, 2022-12-18:
Hey, in the Original Firmware there is a Mode name Edge Switch, this is needed for Cross circuit lamps.

It is Not possible or come it maybe with an new HomeKit Firmware Next Time?

Thanks for answer.

timoschilling, 2022-12-19:
We have that option
[image]
Or if you use a stateless switch its this:
[image]

SunseekerFfm, 2022-12-19:
Thank you very much for that fast support. 
But i dont have this options, but i have a    almost similar variant and it works. 
When you allowing one More question? Any idear why i dont have your proposed options?

Greetings
SunseekerFfm

SunseekerFfm, 2022-12-19:
-[image]

closed 2023-01-27 by github-actions[bot]