mongoose-os-apps/shelly-homekit #1159: A question HomeKit architecture

5xPa, 2022-12-18:
I have upgraded to ios 16.2 without any problems so far. I have recently found within this update. There is an option to upgrade change HomeKit architecture. I was wondering is there any potential issues with this upgrade. 

My setup is 
Shelly 1, 2.5 and UNI which work great so thanks for all the effort and work
Homebridge running plugins for 
Evohome (Honeywell Heating control) 
Texecom Premier Elite Burglar alarm
Dummy switches

timoschilling, 2022-12-18:
That update is only related to HomeKit works inside of Apple Devices not inside the accessories, so it should be save. Some users written that they have updated without issues and my setup works well too.

closed 2022-12-18 by timoschilling

5xPa, 2022-12-24:
I updated to 16.2 and architecture initially all ran Ok on my iPhone 13 and Gen 6 iPad.

I tried to add family members wife has an iPhone SE and iPad Gen 5
Upgraded to 16.2 including architecture upgrade 
Then the trouble started  I couldn't get homekit running on any of these devices. I did the following 
Deleted wifes account from HomeKit
Deleted Homekit
Added back in and send invitation still not showing in HomeKit
Rebooted 3 HomePod Minis and 2 Apple TVs
Wifes HomeKit not showing my home

Called apple they recommended rebooting router etc
Open HomeKit on wifes iPad clicked three dots
There were two "my home" changed selection to the second one HomeKit showed up
Then all sort of devices wouldn't operate.
Some would operate  via HomeKit but not local switch
Web interfaced to device not operating by local switch, selected reboot. Took ages with  a warning of something like time to waiting  60000 Ms to complete
Powered down lighting circuits. after a couple of operations local switches on some devices not operating. Some devices not operating via HomeKit

Deleted the other user user from Homekit wife etc.
So far after about 4 hours everything responding.  
Here's hoping it stays that way.
I note apple have paused architecture update.

Sorry for the long winded notes which are related to HomeKit not mongoose shelly firmware

Have a Merry Christmas and Guid New Year.

Thanks to all those who developed this project

andyblac, 2022-12-24:
HomeKit 16.2 "New architecture" has been pulled by apple, as they have accepted it is too buggy for release to the gen pop. Hopefully Apple will fix their crap fest in the New Year.

5xPa, 2022-12-24:
The problem for me is I can’t find anyway to rollback, iPhone, iPad, HomePod, Apple TVs so I’m stuck with 16.2 and Architecture upgrade

andyblac, 2022-12-24:
You can't that what everyone is complaining about, luckily my setup seams to be ok, although i did the "upgrade" on my M1  iMac and not on my iPhone, as the the upgrade failed when I tried to do via my iPhone, and for some reason that has seemed to make a difference as other have reported the same thing.

timoschilling, 2022-12-24:
I have made it on my iPhone and it works. Some automations are broken but easy to fix them by readding the triggers