mongoose-os-apps/shelly-homekit #1164: Shelly 1 Plus Wifi problems?

bsoener, 2022-12-22:
Hello. have for some time problems with my Shelly plus 1. this I installed half a year ago behind my socket. But recently it keeps losing connection to wifi. I have to log into the Shelly Wlan every time and enter the Wlan address from my Fritzbox and save again. Can it be that the Shelly is broken? Thanks for the help

`558611824 shelly_debug.cpp:231    No log file, sending new entries
558647763 mgos_rpc_channel_ua:167 Incoming frame is too big, dropping.
558657225 mg_rpc.c:314            Shelly.GetInfoExt via WS_in 192.168.178.21:60438
559863991 mgos_rpc_channel_ua:167 Incoming frame is too big, dropping.
560888195 mgos_rpc_channel_ua:167 Incoming frame is too big, dropping.
560949933 shelly_main.cpp:449     Up 560.94, HAP 0/0/16 ns 0, RAM: 206788/193284; st 34; 0.1: st:0 in_st:0 inm:1 ininv:0
563119031 mgos_rpc_channel_ua:167 Incoming frame is too big, dropping.
563544533 mgos_rpc_channel_ua:167 Incoming frame is too big, dropping.
564696736 mgos_rpc_channel_ua:167 Incoming frame is too big, dropping.
565961065 mgos_rpc_channel_ua:167 Incoming frame is too big, dropping.
566516934 mgos_rpc_channel_ua:167 Incoming frame is too big, dropping.
567843559 mgos_rpc_channel_ua:167 Incoming frame is too big, dropping.
568949131 shelly_main.cpp:449     Up 568.94, HAP 0/0/16 ns 0, RAM: 206916/193284; st 34; 0.1: st:0 in_st:0 inm:1 ininv:0
569234322 mgos_rpc_channel_ua:167 Incoming frame is too big, dropping.
569737855 mgos_rpc_channel_ua:167 Incoming frame is too big, dropping.
569811073 mg_rpc.c:314            Shelly.GetInfoExt via WS_in 192.168.178.21:60438
570446647 mgos_rpc_channel_ua:167 Incoming frame is too big, dropping.
570882488 mg_rpc.c:314            Shelly.GetInfoExt via WS_in 192.168.178.21:60438
571908348 mg_rpc.c:314            Shelly.GetInfoExt via WS_in 192.168.178.21:60438
572052374 mgos_rpc_channel_ua:167 Incoming frame is too big, dropping.
572509922 mgos_rpc_channel_ua:167 Incoming frame is too big, dropping.
572932075 mg_rpc.c:314            Shelly.GetInfoExt via WS_in 192.168.178.21:60438
573018773 mgos_rpc_channel_ua:167 Incoming frame is too big, dropping.
573473691 mgos_rpc_channel_ua:167 Incoming frame is too big, dropping.
573932588 mgos_rpc_channel_ua:167 Incoming frame is too big, dropping.
576047820 mgos_rpc_channel_ua:167 Incoming frame is too big, dropping.
576544032 mgos_rpc_channel_ua:167 Incoming frame is too big, dropping.
576951360 shelly_main.cpp:449     Up 576.94, HAP 0/0/16 ns 0, RAM: 204956/193284; st 34; 0.1: st:0 in_st:0 inm:1 ininv:0
577017819 mgos_rpc_channel_ua:167 Incoming frame is too big, dropping.
577461212 mgos_rpc_channel_ua:167 Incoming frame is too big, dropping.
577900179 mgos_rpc_channel_ua:167 Incoming frame is too big, dropping.
578471413 mgos_rpc_channel_ua:167 Incoming frame is too big, dropping.
579360845 mgos_rpc_channel_ua:167 Incoming frame is too big, dropping.
579889639 mgos_rpc_channel_ua:167 Incoming frame is too big, dropping.
580440351 mgos_rpc_channel_ua:167 Incoming frame is too big, dropping.
581005670 mgos_rpc_channel_ua:167 Incoming frame is too big, dropping.
581440616 mgos_rpc_channel_ua:167 Incoming frame is too big, dropping.
582408339 mgos_rpc_channel_ua:167 Incoming frame is too big, dropping.
583405588 mgos_rpc_channel_ua:167 Incoming frame is too big, dropping.
584949750 shelly_main.cpp:449     Up 584.94, HAP 0/0/16 ns 0, RAM: 206264/193284; st 33; 0.1: st:0 in_st:0 inm:1 ininv:0`

pedzp14, 2022-12-23:
same for me on only 1 out of 6 plus1s

timoschilling, 2023-02-24:
@pedzp14 do you have this line in the logs on the shellys that have the problem NOT?
```
565961065 mgos_rpc_channel_ua:167 Incoming frame is too big, dropping.
```

gNeeUs, 2023-04-17:
I am commenting to reopen this case. I have one Shelly Plus 1 on my network that displays the same message in the log continuously, while other devices on the network do not.

mgos_rpc_channel_ua:167 Incoming frame is too big, dropping

I would like to find out what this means.

mikkokam, 2023-04-29:
Shelly i4 doing this as well.

```
826314803050 mgos_rpc_channel_ua:167 Incoming frame is too big, dropping.
826315200063 mgos_rpc_channel_ua:167 Incoming frame is too big, dropping.
826315602333 mgos_rpc_channel_ua:167 Incoming frame is too big, dropping.
826315681844 shelly_main.cpp:449     Up 826315.66, HAP 0/1/16 ns 1, RAM: 203864/194156; st 46; 3.1: svt:3 inv:0 st:0 m:0 lea: 956.526; 3.2: svt:3 inv:0 st:0 m:0 lea: 955.190; 3.3: svt:3 inv:0 st:0 m:0 lea: 956.945; 3.4: svt:3 inv:0 st:0 m:0 lea: -1.000
826315985210 mgos_rpc_channel_ua:167 Incoming frame is too big, dropping.
826316365336 mgos_rpc_channel_ua:167 Incoming frame is too big, dropping.
826316741302 mgos_rpc_channel_ua:167 Incoming frame is too big, dropping.
826317211466 mgos_rpc_channel_ua:167 Incoming frame is too big, dropping.
826317607282 mgos_rpc_channel_ua:167 Incoming frame is too big, dropping.
826318015987 mgos_rpc_channel_ua:167 Incoming frame is too big, dropping.
826318380052 mgos_rpc_channel_ua:167 Incoming frame is too big, dropping.
```

orevron, 2023-11-22:
Having the same Issue here. any updates? did it solved for anyone?

markirb, 2024-06-26:
I saw this also on my Shelly Plus i4 todays.
The reason is that Shelly seems to receive garbage via uart rx. One fix would be to disable uart

closed 2024-07-26 by github-actions[bot]