mongoose-os-apps/shelly-homekit #1166: Expose long push as stateless button in momentary mode

AlexDavidenko, 2022-12-23:
Use case:
By default in momentary mode the long push starts a timer, which is rarely used. 
When using detached mode but the short press still controls the actual relay the short push action takes a long time to perform because of the whole shortcuts logic.

Instead I suggest having an option to either start a timer (default) or expose the long push action as a stateless button with single press only. 
If you need the whole single press, double press, long push you can still use detached mode. 

If exposing long push only in momentary mode you're enjoying both worlds, the relay will trigger faster and you still have long push as a stateless button to control other accessories.

timoschilling, 2022-12-23:
I think you mean something like this https://github.com/mongoose-os-apps/shelly-homekit/pull/1011

closed 2022-12-23 by timoschilling

AlexDavidenko, 2022-12-23:
yeah, perfect. Thanks