mongoose-os-apps/shelly-homekit #1167: Cant reverting to stock firmware shelly

miomia, 2022-12-23:
I have used the ./flash-shelly.py -a and  script to update my shelly's to stock firmware .  I received the error "ModuleNotFoundError: No module named 'zeroconf'" After I tried to install 'zeroconf' with the command "pip3 install zeroconf" I received many errors. How else can I solve the problem? 
[image]
[image]

timoschilling, 2022-12-23:
Looks like you have a problem with your python setup. 

You could also revert via web Ui

miomia, 2022-12-23:
> You could also revert via web Ui
 How can I restore via web Ui ?

timoschilling, 2023-01-09:
[image]

pascal9275, 2023-02-04:
Hello, I have a similar issue.
I have updated a Shelly plus 1pm and wanted to revert it to stock.
I have used the revert from the web UI.
The window indicated it will resend the version 0.12.0 ( I was on 0.13 beta)
The update goes up to 89% and then stops
I made several retry, it always stops at 89 %
Any suggestions?

closed 2023-02-13 by github-actions[bot]