mongoose-os-apps/shelly-homekit #1170: Shelly RGBW2 On/Off Switch Bug

Marfre888, 2022-12-29:
Hi,

I have recently gotten a third Shelly RGBW2. I’ve set this up at RGB+W. So homekit is exposing two different lights. I have a switch connected to GND and I.

It seems that under this setting, only the white leds turn off.

Tbh I do need to do some more research to figure out what is happening, but it seems to me that this is what’s happening.

Will revert back with any more findings. In the meantime if you can replicate this bug please let me know so I know it’s not only localised to my setup.

markirb, 2022-12-29:
This is what’s happening yes. This is expected as it would be undefined behavior to use one switch for multiple lamps. I think this issue was raised before. It is also discussed in the original PullRequest.

You can work around by using it as a detached switch

timoschilling, 2022-12-29:
At this point it’s expected behavior and not a bug. A different solution is already requested somewhere else, so I close this

closed 2022-12-29 by timoschilling

Marfre888, 2022-12-29:
> At this point it’s expected behavior and not a bug. A different solution is already requested somewhere else, so I close this

I don't understand how this is expected behavior. The stock firmware does it without any problem.

I did not see any other requests for this topic, and I still believe this is a bug.

A physical switch should turn all the leds off, and revert back to the previous settings when turned back on. That should be expected behavior, I don't want to have to rely on homekit to turn the leds on/off cause that has proved incredibly unreliable in the past.

markirb, 2022-12-29:
This is expected behavior as it was deliberately implemented that way. Personally I also thought it was better to switch all, but there are also legitimate reasons not to. 

See discussion starting in https://github.com/mongoose-os-apps/shelly-homekit/pull/853#issuecomment-1006638004

Marfre888, 2022-12-29:
> This is expected behavior as it was deliberately implemented that way. Personally I also thought it was better to switch all, but there are also legitimate reasons not to.
> 
> See discussion starting in [#853 (comment)](https://github.com/mongoose-os-apps/shelly-homekit/pull/853#issuecomment-1006638004)

Yes but now we aren't talking about 4 independent white led strips. We are talking about 1 strip which has both RGB and W leds. Why on God's green earth would I ever want to turn off half of the strip? I would either want it all on or all off. My use case is as an under-cabinet kitchen light. If I set it to RGBW mode, it works fine, but only uses half the leds on the strip and it's quite dim. If I set it to RGB+W, the sliders are separated, so I can turn them both on. However, then I need to turn them off using my phone, cause the switch will only turn half of it off which is rather annoying. If it was implemented deliberately this way I don't think there was much thought put into it

Marfre888, 2023-01-05:
> > We are talking about 1 strip which has both RGB and W leds.
> 
> Then you should set the strip not to RGB+W, then it should be RGBW and then all lights are turned off at once. The RGB+W mode is if you have a RGB strip and a dedicated W strip.

Ok, that makes sense then, however in RGBW mode the white led strip is never used. So what’s the difference between RGBW and RGB?

timoschilling, 2023-01-05:
I have tested it with the current 2.11.2 and it works as expected.

Please open the web ui of your rgbw and set him to this settings:
[image]

Is your white channel now on?

If not please open `http://<ip>/debug/log?follow=1` then turn your light off and on again and post the log output from the turn on, it contains `Transition finished, end state:`

Marfre888, 2023-01-05:
> I have tested it with the current 2.11.2 and it works as expected.
>
> Please open the web ui of your rgbw and set him to this settings: [image]
>
> Is your white channel now on?
>
> If not please open [...]

Can you clarify what expected behaviour is?

The screenshot you posted clearly only mentions the RGB strip, but says nothing about the white leds

`308839853451 shelly_main.cpp:443     Up 308839.84, HAP 0/1/12 ns 1, RAM: 27096/20424; st 0; 11.1: sta: off, b: 100, h: 13, sa: 100, ct: 400
308840099433 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.1.6:63029
308840186690 mg_rpc.c:310            Shelly.SetState via WS_in 192.168.1.6:63029
308840193638 shelly_hap_light_bu:197 State changed (RPC): off => on
308840206063 shelly_light_bulb_co:99 Starting transition: [r=0.00 g=0.00 b=0.00 w=0.00] -> [r=1.00 g=0.22 b=0.00 w=0.00], 1000 ms
308840211234 shelly_rgbw_controll:38 Output 1: 0.00 => 1.00
308840215825 shelly_rgbw_controll:39 Output 2: 0.00 => 0.22
308840220267 shelly_rgbw_controll:40 Output 3: 0.00 => 0.00
308840224940 shelly_rgbw_controll:42 Output 4: 0.00 => 0.00
308840317715 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.1.6:63029
308841110388 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.1.6:63029
308841208328 shelly_light_bulb_co:55 Transition finished, end state: [r=0.99 g=0.22 b=0.00 w=0.00]
308844817152 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.1.6:63029
308845375589 mg_rpc.c:310            Shelly.SetState via WS_in 192.168.1.6:63029
308845382590 shelly_hap_light_bu:197 State changed (RPC): on => off
308845394810 shelly_light_bulb_co:99 Starting transition: [r=1.00 g=0.22 b=0.00 w=0.00] -> [r=0.00 g=0.00 b=0.00 w=0.00], 1000 ms
308845399973 shelly_rgbw_controll:38 Output 1: 1.00 => 0.00
308845404550 shelly_rgbw_controll:39 Output 2: 0.22 => 0.00
308845408987 shelly_rgbw_controll:40 Output 3: 0.00 => 0.00
308845413678 shelly_rgbw_controll:42 Output 4: 0.00 => 0.00
308845451774 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.1.6:63029
308845820734 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.1.6:63029
308846398320 shelly_light_bulb_co:55 Transition finished, end state: [r=0.00 g=0.00 b=0.00 w=0.00]
308847999002 mgos_sys_config.c:323   Saved to conf9.json
308848013691 shelly_main.cpp:443     Up 308848.00, HAP 0/1/12 ns 1, RAM: 26872/20424; st 0; 11.1: sta: off, b: 100, h: 13, sa: 100, ct: 400 `

timoschilling, 2023-01-06:
> Can you clarify what expected behaviour is?

Expected behaviour is that only the W LEDs are on and the log shows `Transition finished, end state: [r=0.00 g=0.00 b=0.00 w=1.00]`

> The screenshot you posted clearly only mentions the RGB strip, but says nothing about the white leds

No the screenshot shows RGBW settings, see the mode is set to RGBW on top of the screenshot (but yes there is a bug the lable still says RGB).

Your log shows that you haven't configured like in the screen shot. You still have the saturation to 100% (`sa: 100` in the last log line) and that means it will show a full color and not W.

Marfre888, 2023-01-06:
> > Can you clarify what expected behaviour is?
> 
> Expected behaviour is that only the W LEDs are on and the log shows `Transition finished, end state: [r=0.00 g=0.00 b=0.00 w=1.00]`
> 
> > The screenshot you posted clearly only mentions the RGB strip, but says nothing about the white leds
> 
> No the screenshot shows RGBW settings, see the mode is set to RGBW on top of the screenshot (but yes there is a bug the lable still says RGB).
> 
> Your log shows that you haven't configured like in the screen shot. You still have the saturation to 100% (`sa: 100` in the last log line) and that means it will show a full color and not W.
> 
> Please redo we saturation 0%

> 

> > Can you clarify what expected behaviour is?
> 
> Expected behaviour is that only the W LEDs are on and the log shows `Transition finished, end state: [r=0.00 g=0.00 b=0.00 w=1.00]`
> 
> > The screenshot you posted clearly only mentions the RGB strip, but says nothing about the white leds
> 
> No the screenshot shows RGBW settings, see the mode is set to RGBW on top of the screenshot (but yes there is a bug the lable still says RGB).
> 
> Your log shows that you haven't configured like in the screen shot. You still have the saturation to 100% (`sa: 100` in the last log line) and that means it will show a full color and not W.
> 
> Please redo we saturation 0%

`341632782551 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.1.33:56320
341632964737 shelly_main.cpp:443     Up 341632.95, HAP 0/1/12 ns 1, RAM: 26656/20424; st 0; 11.1: sta: off, b: 100, h: 0, sa: 0, ct: 400
341633782455 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.1.33:56320
341634788602 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.1.33:56320
341635789314 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.1.33:56320
341636120966 mg_rpc.c:310            Shelly.SetState via WS_in 192.168.1.33:56320
341636127950 shelly_hap_light_bu:197 State changed (RPC): off => on
341636140225 shelly_light_bulb_co:99 Starting transition: [r=0.00 g=0.00 b=0.00 w=0.00] -> [r=0.00 g=0.00 b=0.00 w=1.00], 1000 ms
341636145217 shelly_rgbw_controll:38 Output 1: 0.00 => 0.00
341636149649 shelly_rgbw_controll:39 Output 2: 0.00 => 0.00
341636154051 shelly_rgbw_controll:40 Output 3: 0.00 => 0.00
341636158779 shelly_rgbw_controll:42 Output 4: 0.00 => 1.00
341636203087 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.1.33:56320
341636225245 mgos_mongoose.c:66      New heap free LWM: 20320
341636786356 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.1.33:56320
341637145146 shelly_light_bulb_co:55 Transition finished, end state: [r=0.00 g=0.00 b=0.00 w=1.00]
341637785942 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.1.33:56320
341638036326 mg_rpc.c:310            Shelly.SetState via WS_in 192.168.1.33:56320
341638043296 shelly_hap_light_bu:197 State changed (RPC): on => off
341638055453 shelly_light_bulb_co:99 Starting transition: [r=0.00 g=0.00 b=0.00 w=1.00] -> [r=0.00 g=0.00 b=0.00 w=0.00], 1000 ms
341638060439 shelly_rgbw_controll:38 Output 1: 0.00 => 0.00
341638064870 shelly_rgbw_controll:39 Output 2: 0.00 => 0.00
341638069272 shelly_rgbw_controll:40 Output 3: 0.00 => 0.00
341638074044 shelly_rgbw_controll:42 Output 4: 1.00 => 0.00
341638108816 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.1.33:56320
341638786525 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.1.33:56320
341639055068 shelly_light_bulb_co:55 Transition finished, end state: [r=0.00 g=0.00 b=0.00 w=0.01]
341639793133 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.1.33:56320
341640787974 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.1.33:56320
341641110634 mgos_sys_config.c:323   Saved to conf9.json
341641125049 shelly_main.cpp:443     Up 341641.11, HAP 0/1/12 ns 1, RAM: 26444/20320; st 0; 11.1: sta: off, b: 100, h: 0, sa: 0, ct: 400`

Yes this seems to work. However how do I get this setup in Homekit? If I need to open the webui every time I need to change the colour of the LEDs it would make more sense to just use the stock firmware.

AND most importantly, how do I get both the white and rgb leds to turn on at the same time? It’s possible to do it with stock firmware. Is it not possible here?

timoschilling, 2023-01-06:
I think your problem is a bad user interface in the Home App, to get a clear white out of a rgbw lamp you need saturation 0% and that is hard to set in the Home app colorpicker and in the temperaturewheel. If you want that this will be improved open a ticket for the Home app at Apple. Or use different apps, there is for example the Home+ app which has a better user interface for that.

> most importantly, how do I get both the white and rgb leds to turn on at the same time?

Never, there is no case where the color leds and the white leds should be on on full brightness at the same time.
I know the stock firmware can do that because they implemented RGB+W and not RGBW.
What you want in fact the both at the same time and that didn't work or didn't would make sense, because then your led would have more brightness on white then on each other color. 
In a regular RGBW implementation you have four led which are the 100% of your led capacity, but only max ~25% will be on at the same time. Some examples:
1. color red: red is 100% all other off, means 25% are on
2. color green: green is 100% all other off, means 25% are on
3. color yellow: green is 50% and red is 50%, means 25% are on
4. color light yellow (sat 50%): green is 25% and red is 25%, white 50%, means 25% are on
5. color white: white is 100% all other off, means 25% are on
I hope you got it now, the Shelly implementation isn't our benchmark because they implemented RGB+W and not RGBW.
That they implemented RGB+W you can see on the fact that they have a dedicated brightness control for W, in RGBW the W would be controlled by the color wheel.

If you want to have a real RGBW set your Shelly to RGBW, if you want to control / get on all leds at once use the RGB+W mode and live with the tradeoff that you need to control two lamps.

PS: I have seen that you are using homebridge, one of tow reasons while I'm on this project and that this firmware supports the rgbw2 is that the Shelly plugin does a mix of RGB+W and RGBW. For me it feels that you expect exactly that behaviour for our firmware. But sorry the plugin is wrong. As I started with the development of the rgbw2 support I revere engineered two official certificated HomeKit RGBW lamps (Ikea Tradfry and Philips Hue) and they are worked in the same way as the rgbw2 with our firmware works.

timoschilling, 2023-01-06:
Short side comment to your issue #1172, I think our current implementation is targeted for RGBW strips where the W is a could W, and yes that can be improved, I will answer on #1172 if I have some more time.

Marfre888, 2023-01-06:
I understand better your reasoning behind how it is implemented. I understand why you are trying to maintain 25% usage to keep the strip consistent.

However if you take a Philips Hue light strip, it does not maintain 25% through the whole range. The white light (cct) is definitely brighter than say the rgb leds. And the strip is a lot brighter if you set it to a colour that uses the full spectrum vs only red leds for example. If you look at the leds work you can see what they are doing, and that uses RGBWW

While I respect the reasoning behind using only 25%. I don’t understand why you can’t turn on the strip to 100% brightness. The rgb strip is full spectrum so that will do any white light and the rgbw2 can take 100% brightness, so why purposely handicap the light.

timoschilling, 2023-01-07:
We turn the request colors to 100%, but if you request a cold white we only turn on the W strip and not turning only all color leds tow. 
that’s how hsv (hue, saturation, brightness) to rgbw works. And if we would use the rgb leds to produce additional white on a rgbw strip other users will write issues about the dirty white because rgb white is not clean depending on the situation you can see the single colors in the white.

Marfre888, 2023-01-09:
> We turn the request colors to 100%, but if you request a cold white we only turn on the W strip and not turning only all color leds tow. that’s how hsv (hue, saturation, brightness) to rgbw works. [...]

It's only a dirty white if you use low quality rgb leds, and that is why you would add this as an additional mode.. Ultimately you can't cheat nature, and white is made up of a combination of colors. Red, green and blue makes white.

Anyway, I found a way to go about it that I'm happy with, so I'll go with them. Thanks anyway..