mongoose-os-apps/shelly-homekit #1171: Shelly Plus 1PM SSID with apostrphes

Marfre888, 2022-12-29:
Hi,

I have a Shelly Plus 1PM with this firmware flashed. My wifi ssid has an apostrophe (') in it, and for some reason the Shelly won't connect to it. 

It this an issue local to me only? My other shelly devices (not plus devices) on this same firmware have connected just fine to this ssid

Marfre888, 2023-01-05:
Bit of an update.

I’ve managed to connect it to an SSID with an apostrophe (‘). It seems the shelly sees the apostrophe as a different character than what is written.

Connecting it to the stock firmware then loading shelly-homekit fixes the issue.

timoschilling, 2023-01-05:
Is it possible that you made a typo on the first try or had a keyboard layout problem? Because you use different char in issue description and in your last comment.

Marfre888, 2023-01-06:
The first commend was written from my mac and the second from my iPhone. Defo didn’t make a typo, but definitely possible about a keyboard layout issue but I don’t recall if I tried setting it up using my Mac. Either way shouldn’t the device reinterpret the apostrophe to a standard character so avoid this issue?

timoschilling, 2023-01-06:
> Either way shouldn’t the device reinterpret the apostrophe to a standard character so avoid this issue?

Definitely no! What's when you really mean the apostrophe?

closed 2023-01-06 by timoschilling

Marfre888, 2023-01-06:
> > Either way shouldn’t the device reinterpret the apostrophe to a standard character so avoid this issue?
> 
> Definitely no! What's when you really mean the apostrophe?

You’re very quick to close my threads which is incredibly annoying, and probably the reason why this project is full of bugs!

I do want an apostrophe!!! What other keys are there on a standard keyboard that AREN’T a standard apostrophe? Am I creating a new never before used apostrophe? No! I used the same laptop both to setup my wifi and the shelly, and this issue happened. So that is a bug whether you’ll admit it or not.

I could have not bothered, after I fixed my issue, but I came here, took the time to report it, and instead of taking it seriously you closed the thread. Ok whatever, live with the bugs then.

timoschilling, 2023-01-06:
how ever your bug is not reproducible
[image]

timoschilling, 2023-01-06:
When someone writes "My other shelly devices (not plus devices) on this same firmware have connected just fine to this ssid", I already know it's a typo or some other mistake on your site.

Marfre888, 2023-01-06:
The other shelly devices connected because they were first connected through the stock firmware.

At the time I wrote that I didn’t assume this would be how to solve the issue.