mongoose-os-apps/shelly-homekit #1172: Shelly RGBW2: Alternative RGBW Mode

Marfre888, 2022-12-29:
I have an led strip that has both RGB and White leds (RGB+W).

I appreciate all the settings currently implemented as it gives the RGBW2 endless possibilities.

However perhaps the way the RGBW mode works can be slightly tweaked.

Currently, in Homekit, I believe when you are using the Colour wheel (RGB) only the RGB leds are used (which is correct). However, when the Temperature wheel (CCT) is selected only the RGB leds are used. So the white leds remain used.

That said there is plenty of use for these white leds in the spectrum while still maintaining the correct colour temperature.

I also understand there is an RGB+W mode then separates both the RGB and White channels, but then every time I want to choose a different shade of white I need to play around, which isn't in Apple's design policy, and really frustrating.

Therefore I imagine a way to incorporate the white leds with little user input would be to; Choose within the webui whether you have warm white or cool white leds. Then an algorithm would decide how best to blend them in when you are using the Temperature wheel (CCT). That way if your white leds are at 5600K (CW), if you set the slider to 5600k inside of homekit, you instantly get the max brightness of the strip, which would be close to double what the current implementation is. If you set the slider to 4000k, you would get less from the white leds and more from the RGB leds. If you go for colour, you would obviously only get RGB leds. I believe this is how companies like Philips Hue have implemented their leds lights, and it would be amazing seeing it work this way inside of this firmware.

timoschilling, 2023-01-09:
I have some problems with this issue. First you mix a mistake on your side, that the W isn't used on the CCT wheel, with two feature requests.

I read that you use the RGB+W mode and expect that the use of the CCT wheel on the RGB lamp in HomeKit uses the W, but that a wrong assumption. The W in RGB+W is a dedicated lamp and not combined with RGB. You need to use RGBW to use RGB and W in combination.

You want that we use RGB and W to get a brighter white. We will not do that because that will mean that a change on the color temp will change the brightness and that's not what the use intend by changing the color temp.

Your assumptions about the CCT and Color wheel UI in the Home app are not correct I think. The UI suggest that there are two ways to set the color. But on the API level between HomeKit and the Shelly that's not the case, there are only Hue, Saturation and Brightness. So the Home app gives you a cct wheel but calculates that into a Hue / Saturation. That means that our firmware can't act baed on the used color or tmp wheel.

What I take from this issue is support for cold or warm white leds. I have created a new issue for that https://github.com/mongoose-os-apps/shelly-homekit/issues/1182

closed 2023-01-09 by timoschilling

Marfre888, 2023-01-09:
While respecting your side of the argument. I just want you to look at Philips Hue light strips operate.

Note ta Philips Hue light strips uses 3x leds (1x RGB, 1x Warm White, 1x Cool White)
Camera has locked exposure

Light Strip LEDs

https://user-images.githubusercontent.com/63603736/211353009-4c60b987-e601-452b-be59-773a1624e4bc.mp4

Light Strip Light on Wall

https://user-images.githubusercontent.com/63603736/211356050-bbf39693-8ab3-490a-b8aa-64095e5dc192.mp4

As you can clearly see, the light the led is producing isn't consistent throughout the full range at all (at the same brightness level). In fact if you go on the Philips Hue website you'll quickly realise this is clearly written in their specifications. 1030lumen @4000k and 570lumen @2700k.. Reason being, that 4000k is a combination of 3200k and 5600k (so at this stage the lightstrip would be using 2 of it's 3 leds. If you look even closer you'll realise that they are cheating by using a Green/Red leds from the rgb strip to achieve the desired colour temperature and if this implementation works for Philips Hue, there is not reason why it shouldn't also work here

Marfre888, 2023-01-09:
@2000K
[image]

@4000k
[image]

@6500k
[image]

timoschilling, 2023-01-10:
You can't compare the functionality of a RGBW with a RGBWW.

How ever, if you think a Philips Hue is better, be happy with it.