mongoose-os-apps/shelly-homekit #1177: Power Meter shows in HomeKit

FrankBla, 2023-01-04:
Hello,
i miss the Power Meter Visualisation in apple HomeKit. Shelly2.5; shelly1PM; and shellyRGBW2 has a build in Power Meter. That is shown on specific web page of the shelly. There is actual and a total wastage displayed. It where a cool Function if this 2 power meters was shown in home kit like the the temperatures from shellyaddon.
Can you please code this, or thinking about that?
Greetings from Germany, Frank

timoschilling, 2023-01-04:
The power consumption is already available in HomeKit. You need Apps which supporting the Eve power characteristics, the Apple home don’t do that. The RGBW2 didn’t support it at the moment

closed 2023-01-04 by timoschilling

FrankBla, 2023-01-05:
Good morning,
now if you say this ì‘ve been open eve and: tadaaa there is shown the used power. Ok, so long i didnt use eve. I‘ve using this Firmware while i like the appleHomekit. Now i must hope, that apple will this build in in his Software or i must use another like eve.
Thank you for your fast answer. This is a very good Support from you.
Have a nice day, Frank

Frank Blaser | Tulpenstrasse 13 | 88317 Aichstetten

timoschilling, 2023-01-05:
A short explanation: Apple HomeKit is the framework which controls the accessory state, executes the automatisations and manages the communication between iOS and the accessories (Shelly). The Home App (which I think you mean with "apple Homekit") is just a reference implementation from Apple. Back in the days where HomeKit has started it has only worked with third party apps, Apples Home App is started 2 years later. HomeKit is designed to be extendable, so Apple predefined characteristics (state: on/of or brightness: 90%) and third party apps can extend that with own characteristics (power consumption). The Home App didn't support everything, even not all Apple defined features of HomeKit are supported in the Home App.