mongoose-os-apps/shelly-homekit #1180: Shelly 2.5 nicht synchron

iStump, 2023-01-07:
Hallo zusammen

Leider ist mein Englisch zu schlecht um mein Problem zu beschreiben, daher versuche ich mein Glück in Deutsch. 

Ich habe meine Rollläden mit Shelly 2.5 Smart gemacht, dies Funktioniert soweit eigentlich Perfekt. Ich kann alle Rollläden auf und zu machen wie gewünscht. 
Allerdings will ich meine Rollläden oft nur Kippen also im ios Home App nur 1% öffnen, wenn ich solche Szenen Aktiviere funktioniert auch dies Perfekt, der gewünschte Rollladen stellt sich dann gekippt auf 1%, aber irgendwie weiss es dann mein ios Home App und die Websteuerung nicht. Es bleibt bei geschlossen (0%) sprich wenn ich dann irgendwann denn Rollladen ganz schliessen will passiert nichts da er denkt er sei bereits zu. 

Erwarte ich da zu viel? Oder habe ich eine Einstellung falsch? 

Sorry nochmal für das Deutsch, hoffe jemand kann mir trotzdem helfen.

Im Screenshot findet ihr meine Wandschalter und Screenshots. Während den Screenshots aufnahmen war der «Rollladen Büro» auf 1% mit einer Szene Aktiviert.
[image]
[image]
[image]

timoschilling, 2023-01-07:
Deutsch ist kein Problem!

mich müsste noch mal in Ruhe nachlesen aber wenn ich mich richtig entsinne brauchen wir etwas Toleranz damit ein Rollo als geschlossen angezeigt wird auch wenn die Berechnung sagt es ist noch nicht ganz zu, den die genaue Position kann nur mit einer Berechnung geschätzt werden. 

Du könntest es mal mit 2,3,4,5 oder 6% versuchen. Irgend wann wird nicht mehr abgerundet. 

Ich werde in den nächsten Tagen aber auch noch mal nachschauen wie es genau ist

iStump, 2023-01-07:
> Deutsch ist kein Problem!
>
> mich müsste noch mal in Ruhe nachlesen aber wenn ich mich richtig entsinne brauchen wir etwas Toleranz damit ein Rollo als geschlossen angezeigt wird auch wenn die [...]

Danke für die Super schnelle Antwort. 

Dies mit mehr % habe ich versucht und funktioniert einwandfrei, allerdings ist es so dass ich bei den 1% denn Rollladen auf ca 45 Grad geöffnet habe, also Heller Raum aber man sieht nicht rein. Bei 2 % ist er bereits 90 Grad offen und man sieht durch, hoffe du verstehst wie ich das meine.

x23piracy, 2023-01-13:
Hi,

das größte Problem ist halt die Laufzeit, der Shelly weiß nur Anhand der Dauer (Stromaufnahme oder eben nicht) ob der Rolladen ganz oben oder unten ist. Mal angenommen die gesamte Laufzeit ist 15 Sekunden dann entspricht eine Sekunde 6,66~ % der gesamten Öffnung bzw. Schließung des Rollos.

1% würde dabei einer Laufzeit von 150,15 Millisekunden betragen, allein durch die Relaiseträgheit ist das ein Tanz auf Messers schneide.

Gruß X23

iStump, 2023-01-13:
> Hi,
>
> das größte Problem ist halt die Laufzeit, der Shelly weiß nur Anhand der Dauer (Stromaufnahme oder eben nicht) ob der Rolladen ganz oben oder unten ist. Mal angenommen die gesamte Laufzeit ist [...]

Hallo X23 

Danke für deine Erklärung. Seit @timoschilling’s Nachricht habe ich meine Szenen Angepasst für das Spontane Lüften auf 3%. Jetzt öffnet sich der Rollladen, also macht ein spalt auf zwischen Boden und Rollladen, was nicht Perfekt ist, aber wenn ich dann die nächste Szene aktiviere schliesst er sich wie gewünscht. Von dem her ist mein Problem soweit eigentlich gelöst.

timoschilling, 2023-01-13:
Ich schließe den Issue da eine Lösung gefunden wurde und das eigentliche Problem nicht behebar ist.

closed 2023-01-13 by timoschilling