mongoose-os-apps/shelly-homekit #1183: Shelly Roller Shutter Mode not correct behaviour

pasqualito79, 2023-01-10:
Hello, I flashed and calibrated a 2.5 Shelly. Now with homekit, I'm not sure if the stop function acts like it should. Example: I click on the icon in homekit to close. Then I click again. Now it stops (it says "will open"...but it just stops). Then clicking again it opens. 
So far, this behavior probably works as it should (with the small glitch that it shows the wrong action for a second). 
But now, I do the same with Siri. When I say close, it closes. When I say open (when it still closes)  it just stops. Then, when I say close (or open) it just does the opposite sometimes, but not always. 
Therefore I'm not really sure that the stop action is really intended like that. On the wall I have two buttons. One down, one up. I configured the Shelly in momentary mode.

timoschilling, 2023-01-10:
Does it work as expected if you control it via Home app?

pasqualito79, 2023-01-10:
Yes, sorry, with "homekit" I meant I controlled it via Home app. I'm not sure if this is how it should be: 1. click = open, 2. click = stop, 3. click = close. 
If this is how Homekit always handles roller shutters, then it's like it is and works reliable.

But with Siri, it's very frustrating. I need to say commands two or three times to work (Siri just says "something is wrong"). 
Back to the bahaviour: 1. Siri "open shutter" = open. 2. Siri "stop shutter" = stop. 3. Siri "open shutter" = close. 
So for me it seems like Siri isn't really listening to the exact command but just cycles through the commands like when clicking the shutter symbol in the Home app.

Can you confirm this bevahiour? If yes, it would be interesting what your influence in the shelly firmware is or if this is just an Apple Siri bug.

timoschilling, 2023-01-10:
That's an Siri problem. The Shelly has no knowledge about open, stop or close, it knows only a target and a current position. If you say Siri "stop" it sets the target position to the current position and so on. So Siri does something wrong there.

closed 2023-01-10 by timoschilling