mongoose-os-apps/shelly-homekit #1184: Shelly is Assen to HomeKit but won’t connect

rmkromkamp, 2023-01-12:
I have a Shelly 2.5. It has the newest HomeKit firmware. It’s connected to the WiFi. I van control it on the web in safari. It connects to HomeKit. But it’s stil saying it’s offline

timoschilling, 2023-01-12:
Where do you see it as offline?

rmkromkamp, 2023-01-12:
Only in HomeKit itself. But I could just add it into HomeKit.

rmkromkamp, 2023-01-12:
Installed the last firmware on the Shelly after I found this failure but didn’t fixed it. Have an Apple TV

timoschilling, 2023-01-12:
Is the Shelly the only HomeKit devices you have?

rmkromkamp, 2023-01-12:
No I have Hue, Aqara, Eufy and HomeBridge. Work all perfect. It’s only this Shelly

timoschilling, 2023-01-12:
Try to reset the HomeKit connection and reconnect them

rmkromkamp, 2023-01-12:
Ive rebooted the Apple TV and I’ve removed the Shelly HUB from the Home app. Then connected it again into HomeKit. Adding the Shelly HUB and the 2 switches worked again fine. Only it’s still disconnected in the Home app.

A-vesalius, 2023-01-15:
I have had the same issue with several Shelly 1L's I bought. Still unuseable.

https://github.com/mongoose-os-apps/shelly-homekit/issues/1099#issuecomment-1309706859
https://github.com/mongoose-os-apps/shelly-homekit/issues/1099#issuecomment-1361619364

rmkromkamp, 2023-01-18:
I moved back to the normal Shelly firmware. But wasn’t even able to add it to HomeKit with HomeBridge. Assen the HK firmware again and added the Shelly back into HomeKit. Adding works fine, but it’s stil offline when it’s added to HomeKit. I’m stil able to control it in the web-interface. Driving me crazy

timoschilling, 2023-01-18:
Have you try to remove them from HomeKit and not only resetting them via Ui?

rmkromkamp, 2023-01-19:
Yes multiple times. With resetten and without resetten

timoschilling, 2023-02-03:
How is your wifi setup? Do you have multiple access points? Is the non working Shelly on a different AP, then the other?

rmkromkamp, 2023-02-03:
I have the solution!! I changed the Mesh WiFi routers into ‘Access Points’ and now everything works fine. That must have changed by some update. But ALL other devices still worked. Was only the Shelly that had a problem. But problem is solved now !! Thank you all for your help

closed 2023-02-03 by timoschilling