mongoose-os-apps/shelly-homekit #1186: Shelly 2.5 stops working for automations and wall switch while accessible by webif

x23piracy, 2023-01-12:
Hi,

i have 5 Shelly 2.5 to my mothers flat and one of them had the following problems 3 times in the last weeks/months it seems that the distance between this problem seems to be similar (2-3 weeks).

Automations for opening and closing (Homekit) stop working and also the wall switches can't seem to move the blinds, what i tried so far was to see if it responds to the webif and yes it does and i tried to recalibrate, which works and after that also the wall switches are back in function.

What the hell is that? :-)

All that Shellys have the latest FW:
Version: 2.11.2
Build: 20220812-225358/2.11.2-g1b146b4-master 

Any idea?

Best Regards X23

5xPa, 2023-01-13:
I have had similar issues after installing the Homekit Architecture upgrade. Hopefully  apple will resolve the issue.

x23piracy, 2023-01-13:
How should an HomeKit Architecture affect the Wall switches? Also it's 1 of 5, i don't think this has anything todo with Homekit itself.

Best Regards X23

x23piracy, 2023-01-13:
Same issue: https://github.com/mongoose-os-apps/shelly-homekit/issues/1157

5xPa, 2023-01-13:
All I can say is I have more than a dozen with the firmware installed for
in excess of 100 days with no problem
and immediately after Homekit architecture upgrade 4 off Shellies would not
control via the wall plate  switch.
Occasionally the problem returns requiring to power off and on the circuit.
Have you installed the Architecture upgrade.

timoschilling, 2023-01-13:
Does you use the normal switch mode or do you use detached switches?

5xPa, 2023-01-13:
Timo In my case HAP Type "switch" input mode "edge" also have some devices set to "momentary".
As I mentioned earlier pre Homekit architecture upgrade I had shelly 2.5 an 1 running perfectly for greater than 100 days.
Every now and again the wall switch will not control the light. Requiring power Off. It certainly appears to be related to Architecture upgrade.

x23piracy, 2023-01-13:
Hi,

in have the classic separated switches (Toggle).

Best Regards X23

x23piracy, 2023-01-13:
Is it possible to debug things when this happens next? Tasmota offers some kind of shell where you can see things that happen, is there anything similar with the HK FW?

FlorianTDK, 2023-01-13:
I’ve got two Shelly2.5. Both for roller shutter. And one of them (always the same) has that problem. I’ve got to recalibrate it every time. It can happen two times a day, then nothing during weeks and randomly do it again.
Since around two years I’ve got them I’ve always had this same issue. And always with the same Shelly2.5 (I tried to hard reset it, to plug it in back, and so on… but nothing does fix it)

x23piracy, 2023-01-13:
Is the general function of the HK FW based on the substructure of the original FW? OR is it working on it's own? I am asking to understand what the cause of the issue might be.

x23piracy, 2023-01-13:
One idea, i am not tracking the temperature sensor, but how to the shellys react if a risky temperature is reached do they switch in some security mode were it switches of the relays?

x23piracy, 2023-01-13:
Interesting: https://github.com/mongoose-os-apps/shelly-homekit/issues/760#issuecomment-932727118
Is this some kind of long time issue?

MiBrus, 2023-01-27:
I have the same issue.

masip85, 2023-03-02:
Same issue here. I've got 6 different shellys and this firmware works perfectly for 5 of them. But for one of my shellies, I can calibrate well, and work fine during 4 or 5 pushes. But at some point, it becomes unresponsive. Can't cancel continuous movement with one push. And both buttons and UI becomes completely unresponsive. It can be during a few seconds, or even indefinitely until new calibration.  

If I revert to stock firmware, It works perfectly. So I guess hardware is right. 

Another issue I have seen is that is loosing the connection quite often, even I don't see in the logs, I can see the red flag at the UI for a few seconds. 

Logs doesn't  show any error or disconnection.

```
28878402 shelly_main.cpp:443     Up 328.86, HAP 0/0/12 ns 0, RAM: 28176/21832; st 42; 4.1: c:1 mp:146.89 ip:5.00 mt_ms:14588 cp:18.42 tp:50.00 md:1 lmd:1
329658167 shelly_hap_window_c:335 WC 1: Cur pos 23.22 -> 23.90, P = 147.56
329742152 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.1.26:35708
330458250 shelly_hap_window_c:335 WC 1: Cur pos 28.70 -> 29.39, P = 147.88
330574494 mgos_http_server.c:180  0x3fff5394 HTTP connection from 192.168.1.26:40526
330744807 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.1.26:35708
331258274 shelly_hap_window_c:335 WC 1: Cur pos 34.19 -> 34.87, P = 147.38
331743012 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.1.26:35708
332058293 shelly_hap_window_c:335 WC 1: Cur pos 39.67 -> 40.36, P = 147.54
332756765 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.1.26:35708
332858381 shelly_hap_window_c:335 WC 1: Cur pos 45.31 -> 45.84, P = 147.12
333457755 shelly_hap_window_c:346 WC 1: Tgt pos 50.00 -> 49.95 (fixup)
333463305 shelly_output.cpp:63    Output 2: on -> off (moving)
333469171 shelly_hap_window_c:326 WC 1: State: moving -> stop (23 -> 24)
333713509 mgos_sys_config.c:323   Saved to conf9.json
333721346 shelly_hap_window_c:326 WC 1: State: stop -> stopping (24 -> 25)
333735577 shelly_hap_window_c:326 WC 1: State: stopping -> idle (25 -> 0)
333747726 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.1.26:35708
334745169 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.1.26:35708
335581354 mgos_http_server.c:180  0x3fff5394 HTTP connection from 192.168.1.26:40540
335744935 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.1.26:35708
336380394 shelly_input.cpp:76     Input 2: change (state 1)
336386899 shelly_hap_window_c:346 WC 1: Tgt pos 49.95 -> 0.00 (ext)
336392787 shelly_hap_window_c:326 WC 1: State: idle -> move (0 -> 20)
```

masip85, 2023-03-02:
> 

I agree

closed 2023-03-10 by github-actions[bot]

FlorianTDK, 2023-04-03:
Any update on this issue?
Thanks