mongoose-os-apps/shelly-homekit #1187: Doorbell switch doesn't ring when Detached

Macrox14, 2023-01-15:
Hello, I have installed my Shelly 1 Plus and when I press the physical switch my doorbell rings (input mode Momentary).

But when I changed it to Detached, the physical switch stop working.

[image]

Am I doint something wrong?

This is my cable
[image]

PeterVanDerSanden1975, 2023-01-15:
Same schema here but other inputs:
[image]

Note: hap type doorbell does not work. Motion detect works well.

Macrox14, 2023-01-15:
@PeterVanDerSanden1975 I’ve tried your schema and same issue: the digital switch works, but the physical doesn’t

PeterVanDerSanden1975, 2023-01-15:
HomeKit should trigger the physical when the input is enabled:

[image]

it is in Dutch, sorry. Let your HomeKit couple switch and physical.

good luck.

timoschilling, 2023-01-15:
Yes you do something wrong. What you try to do isn’t supported. The detached doorbell mode is only to signal a push on the button in HomeKit as doorbell ring, not to control the doorbell it self. You need to wire the doorbell as bevor you installed the Shelly and then you wire the Shelly input in parallel to the doorbell and not in series like it’s now. 

Maybe someday we will support your wireing as well. There is already a feature request for that.

closed 2023-01-15 by timoschilling