mongoose-os-apps/shelly-homekit #1189: Working, Back to Factory Firmware, Now Request Timed Out Setting Up Homekit

nesuser2, 2023-01-16:
I had these working well. Went back to factory firmware for a period and now when I go back to the Mongoose firmware it will not allow me to setup homekit firmware. When I choose setup it comes up request timeout after about 10 seconds. I've tried to tweak things in the network thinking I have put a road block in my way but I can't seem to crack the issue and get it added back to homekit. 

Some of the next steps I can try are to connect a fresh shelly up and go through the normal process to get it online. 
^^ Take that same shelly to another house or router to eliminate my network

Any suggestions welcome..seems like something I have to be doing on my end!

nesuser2, 2023-01-17:
I could toggle the AP on some of them and get it to work. On others I would have to turn on the AP and connect to it to get the HomeKit part to start working. Weird…perhaps it’s been too long since I set these up and I forgot too much.

timoschilling, 2023-01-17:
Sometimes it needs some tries to successfully peer with HomeKit. I also have some Shelly's which are fast and some which needs retries. 

Are all connected now?

nesuser2, 2023-01-17:
Mine were definitely not responding at all before, I tried over several weeks at various times and never had any luck getting any of them to respond.

closed 2023-01-17 by timoschilling