mongoose-os-apps/shelly-homekit #1194: sensor data precision lost

rs38, 2023-01-19:
Shelly plus 1PM:

I just did the switch to Homekit firmware and added it to Homekit, which was seamless and easy and works fine!!

However I noticed that the power (instant and sum) sensor data via Homekit changed in precision and some are lost (e.g. volts).
Native Shelly connection was providing 1 decimals for power and and now passed through to HK it's only integer.
Also energy sensor data is way less in precision.

Is this behaviour intentionally or a limitation of HK?

(Philips Hue bridge with Matter beta shows a very similar reduction in precision compared to native connection)

timoschilling, 2023-01-19:
I tweeted the UI it will show one digit in the next release. But in HomeKit we are limited to int values. Volts will come in the future.

closed 2023-01-19 by timoschilling

rs38, 2023-01-19:
Why are you limited int in Homekit?
If I query power, energy and more from connected _Eve Energy_ device in Homekit Automation, I get floats with many digits. 
Also the state of the power switch is "Ja" instead of "1"...
May be they cheat with strings?

timoschilling, 2023-01-19:
>> But in HomeKit we are limited to int values.
>
> Why are you limited to int values in Homekit?

We don't do that, the specification does it. HomeKit has no official support for W or Wh (V too), but HomeKit supports custom values, that's what Eve does. There is no official specification about what Eve does. So we depend on reverse engineered specifications. That specification says that it is an integer value. Maybe Eve has a newer specification which now supports floats. I have tested our implementation rewritten as float and the Eve App still shows an integer value.

> Also the state of the power switch is "Ja" instead of "1"...

Where do you see that? Can you post screenshots from an Eve and from a Shelly.

rs38, 2023-01-20:
[image]

This outputs
`1;Ja`

timoschilling, 2023-01-20:
To the `1;Ja` problem, it's a bug in shortcuts. It took me 3-4 hours to investigate that and to find an evidence. But here it is: The Shortcut app has different behaviour, depending on some constraints which didn't make sense. First fact is all Shelly are providing a boolean value for the power state, same as Eve, Meross and Koogeek (that's are the devices I have access to for testing) power plugs does. The value in HomeKit is a boolean, in a computer system a boolean is represented as a 1 bit number, that's why you sometimes get a 0 or 1 as value. What does Shortcuts does it's translates this boolean in to a string (`yes`/`no` or `ja`/`nein`). To know what it's the context right representation of a boolean or integer in other cases is, it uses the accessory (HomeKit representation of a shelly) category and not the service (HomeKit representation of one switch configuration in the shelly) type. That didn't make sense, for example you have a Shelly 2.5 out1 configured as switch and out2 configured as garage door opener (this combo isn't implemented at the moment, but would work), then Shortcuts never could have the right translation for each service, because it depends on the accessory type. During the test I also found that a Meross 4x power plug which is original certificated also produces `0`/`1` and not `no`/`yes`.

If you want to discuss more about that power state thing, open a new issue, discussing it here mixed with the float thing is a irritating.

A update to the power values will follow.

rs38, 2023-01-20:
thanks for your investigations!

the Shurtcut/Automation part within Homekit App was crashing countless times, not wondering there are more bugs.
It's also a mess to find the meaning of the variables, all with the same name "eigene".
Not mentioning the error prone "user friendly" 🤣, globalized-language, touch-programming with no text editor to get the values finally in some cloud store.

Update of power values in HK means you will follow up with some findings?

timoschilling, 2023-01-21:
little pre information: a HomeKit Characteristic like the current power consumption, has 3 range values, a min value, a max value and a step value.

I found the mistake I made ~2 years ago, as I have made the initial implementation of the power stuff. I have tried to implement it as float then, but I made a mistake and set the step value to 1.0, that's resulted in homekit rounds the values up to an integer. I have fixed that now.

Here is the PR for that feature: https://github.com/mongoose-os-apps/shelly-homekit/pull/1197
Tomorrow I will provide some builds for all power measurement Shellys, in that PR.

timoschilling, 2023-01-21:
prerelease build are now linked in the pr https://github.com/mongoose-os-apps/shelly-homekit/pull/1197

rs38, 2023-02-07:
@timoschilling 
for the decimals regarding power value I commented in your PR already, seems the sensor doesn't really keep up with that precision.
but:
the amout of energy value reported via Homekit is also not working as expected. Checked with Eve App and Homekit Shortcut it reports only an Integer = 0 which is probably kWh or a bug. 
Is it expecting Wh?

timoschilling, 2023-02-07:
@rs38 I still on it, there is a problem and I'm not sure what it is. The sensor itself can 3 decimal digits. I will update the PR if there is something new.

rs38, 2023-02-07:
how can I help/assist? 

what are the units you are speaking of?

let me think loud:
According to https://www.belling.com.cn/media/file_object/bel_product/BL0937/datasheet/BL0937_V1.02_en.pdf the accuracy of the power cannot be better than 1W. Even 1 decimal do not really make sense for that particular device.
Depending on how often the power is retrieved internally like ~1Hz, 3 decimals of Wh (=3600Ws) make sense and this is what the original Shelly and custom firmware shows on the decvice's web UI.

But at least the Eve App is only showing kWh with 2 decimals, which would be a minimum of 10Wh. But how would that fit into sth. which looks like 16bit Uint plus 1 decimal?
UFloatCharacteristic kHAPCharacteristic_EveTotalConsumption, 0.0f, 65535.0f, 0.1f,

if you gather energy in Wh with 3 decimals, you will end up having kWh with 6 decimals.

Are there any reads for a newbie like me? Or even better: a test suite to play with virtual sensors and learn from the behaviour in the Eve app and Homekit "layer"?

timoschilling, 2023-02-07:
I need time to recap every findings and retest some things. Will come back with an update then.

PR #1196 (2023-01-19): improve power meter
closes #1194