mongoose-os-apps/shelly-homekit #1200: Shelly 2.5 - failed to flash OTA - no new update

Olavve, 2023-01-23:
Upon running the OTA command, the following status is displayed:
{"status":"updating","has_update":false,"new_version":"","old_version":"20221027-092056/v1.12.1-ga9117d3"}
The update is not performed.

timoschilling, 2023-01-24:
That can happen sometimes, please do some retries. Can you Shelly access the internet or do you have some restrictions for him?

Olavve, 2023-01-24:
Thx for the reply. I did try it a few times already including previous Shelly reboot. 
Access of the UI of the Shelly via the ip address continues as normal. No restrictions after attempted flashing.

Olavve, 2023-01-25:
Hi Timo, I tried the flash procedure possibly 20 times using the direct address for Shelly 2.5 and the general OTA address. With soft reboot of the Shelly and hard reboot (taking off mains). 
The Shelly firmware remains in place and fully operational. 
Any other tricks to convince the Shelly to flash to Shelly-HomeKit?
PS: the flashing worked fine on my Shelly 1

timoschilling, 2023-01-25:
Do you have updated the original firmware to the latest version?

Olavve, 2023-01-25:
Yes I did.
20221027-092056/v1.12.1-ga9117d3

closed 2023-03-05 by github-actions[bot]