mongoose-os-apps/shelly-homekit #1204: Shelly’s without function on Network/Switch

MatthiasNeuss, 2023-01-24:
I have 20 Shelly 2.5 and Shelly 1 PM. I have previously connected them via Homebridge, rarely problems. 
Since I switched to Shelly HomeKit, I have frequent problems, with both models.

The devices disappear from the WLAN. They cannot be found on network, are neither accessible via WLAN, nor can the connected switch be operated. The connected lamps remain deactivated.
When switched off from power in the house, the devices are then accessible via the Shelly WLAN/Accesspoint mode. The home WLAN is then deactivated. After reactivating the wifi mode, the devices work again. 

Firmware version: 2.11.2

MatthiasNeuss, 2023-02-10:
I was able to specify the problem a bit more.
After 2 weeks, 7 Shellys (different models) are no longer accessible via WLAN. At least I can still switch on the lamps with the wall switch on 2 Shellys. With the others, the wall switch no longer works either. 
Also the trick with the power off did not work everywhere. I had to remove 3 Shellys completely and reset them with the reset button on the relay. 
Do you have any idea what the problem is?
Do you need more information?

MatthiasNeuss, 2023-02-21:
I have the problems with the Shelly 2.5 and with the Shelly 1 PM. 
The Shelly 2.5 are used as wall switches, sockets relays, but also as a shutters. Problems but identical across all variants of use.

It is just really very strange that then also the hardware-switches on the wall no longer work. If only HomeKit was no longer accessible, that would be annoying, but not so critical. Currently, however, all the devices are then also no longer controllable with the manual switch.

d4rkmen, 2023-08-10:
Hi, this is very looks like HW problems. Have devices ventilation is well? When you touch it, is it hot?

seba-78, 2024-02-13:
I have the same problem with a Shelly plus 1. Also several times it got in a not reachable state. Sometimes after some hours without interaction it comes back, but is then in a reset state.