mongoose-os-apps/shelly-homekit #1207: Shelly-1 loses HomeKit Configuration

Cibernello, 2023-01-27:
I installed 2 Shelly-1 on my network. Both configured as switch, both running 2.11.2.

both of them - from time to time - seem to lose HomeKit configuration. They are no longer responding via Home App. The only way to recover the situation is accessing them via browser and launching a soft reboot. 

This is what I see on device local webpage when the switch no longer responds to Home App:

HomeKit Settings
Paired: yes
Connections: 0 pending 0 active 12 max

Any suggestion?

timoschilling, 2023-01-27:
Do you have a HomeHub (HomePod or AppleTV) in your setup? Did you made the HomeKit infrastructure update with 16.2 or 16.3?

Cibernello, 2023-01-28:
I have 1 AppleTV and a HomePod stereo pair updated to the latest software release (16.3)

1yoda, 2023-01-28:
I've had the same problem for 2-3 days.  some shellys lose connection to HK and it says "no responce" from the web i can access the shellys and it is: 

HomeKit Settings
Paired: yes
Connections: 0 pending 0 active 16 max

sometimes they came back in HK, and then do not response again. I do have 14 shellys, but there are same 3-4 shellys that are having this problem all the time

I have ATV with 16.2 (Had updated the Homekit infrastructure, two hours later Apple says that it was better not to do this update, but it was already too late.) + AHP mini with 16.3

timoschilling, 2023-01-28:
@1yoda @Cibernello do you have multiple wifi access point or just one?

Cibernello, 2023-01-28:
I have a netgear Wi-Fi router + netgear mesh range extender. Both broadcasting 2 Wi-Fi networks @ 2,4 and 5ghz. 2 separate SSIDs.

timoschilling, 2023-01-28:
It can be a problem with your wifi setup. Passible the communication between the router and the mash and/or the communication between your 2.4G and 5G networks isn't working well.

1yoda, 2023-01-29:
> @1yoda @Cibernello do you have multiple wifi access point or just one?

I do have multiple wifi access points. Ist a FritzBox with 2 mesh extenders.

All shellys are on 2,4G network. 

But today all shellys are responding well 🤷‍♂️

1yoda, 2023-01-29:
Upd:
One shelly is not responding again 😢

BrennanDee, 2023-01-30:
Same problem here. 
Updated my Shelly 2.5 to 2.11.2 and Apple TV (A1625) to 16.3 (20K650). Can establish a connection to Home App but immediately switches to "No Response" with the Shelly displaying "Paired: yes, Connections: 0 pending, 0 active, 12 max". 
All devices are on a single 2.4G network with one single AP.

bolagnaise, 2023-02-03:
Experiencing same issues since 16.2 on new HomeKit infrastructure with Shelley 2s and 2.5s. Something has changed. If I reboot all my devices they stay connected for 1-2 days then they lose connection with HomeKit and appear as no response. Devices are on a seperate 2.4Ghz IoT WLAN locked to a single AP. Previous to new infrastructure they were rock solid.

timoschilling, 2023-02-03:
The difference between the old and the new HomeKit infrastructure is, that in the old system each device (iPhone, iPad, HomePod, etc) holds it's own connection to the accessories (Shelly), in the new system one HomeHub (HomePod, Apple TV) holds all connections and tunnels that information to the devices.

If you start pairing an accessory to HomeKit first the iPhone connects to the accessory and then handover to the HomeHub. And I think there is the problem. The iPhone can reach the accessory but the HomeHub not. That could have multiple issues. 
1. A network issue
    1. Some routers didn't forward mdns traffic correct, so the HomePod can see the accessory
    2. If you have multiple WiFI's or dedicated IoT networks, there can also be connection issues
2. Apple TV as HomeHub didn't work well, there are bringing sometimes trouble
    1. You could try to reboot them, we had already that issue

timoschilling, 2023-02-03:
For the record: a similar issue was solved as network problem https://github.com/mongoose-os-apps/shelly-homekit/issues/1184#issuecomment-1415738922

closed 2023-03-07 by github-actions[bot]