mongoose-os-apps/shelly-homekit #1208: HAP service type for heater

paulolesgazz, 2023-01-28:
Hi,
I would like to use some shelly 1 to manage heaters in my home. Can you add heater in HAP service type ?
Thanks

5xPa, 2023-01-28:
I would also make a request. I use Uni modules connected to texecom Alarm system. Would be great if there were  Hap
 types Alarm on (system Armed)
           Alarm Active (Intruder detected)

timoschilling, 2023-01-28:
This are double requests of https://github.com/mongoose-os-apps/shelly-homekit/issues/895 and https://github.com/mongoose-os-apps/shelly-homekit/issues/552

closed 2023-01-28 by timoschilling

paulolesgazz, 2023-01-28:
Reading of temperature is done by Eve Weather module. I can put this module anywhere in room (without wire). Shelly 1 will be used only as relay and put directly in socket of heater.

timoschilling, 2023-01-28:
ok https://github.com/mongoose-os-apps/shelly-homekit/issues/895 requested support for thermostat and not heater. I reeded the HAP specification and a heater needs the current temperature, so we can add a heater, but it would not work for your case.