mongoose-os-apps/shelly-homekit #1214: Hello, I have a similar issue.

timoschilling, 2023-02-05:
Hello, I have a similar issue.
I have updated a Shelly plus 1pm and wanted to revert it to stock.
I have used the revert from the web UI.
The window indicated it will resend the version 0.12.0 ( I was on 0.13 beta)
The update goes up to 89% and then stops
I made several retry, it always stops at 89 %
Any suggestions?

_Originally posted by @pascal9275 in https://github.com/mongoose-os-apps/shelly-homekit/issues/1167#issuecomment-1416869708_

timoschilling, 2023-02-05:
I created a new issue, because I think you have a different problem, the context of revert isn’t enough.

timoschilling, 2023-02-08:
It's normal that the progress didn't run to 100%, because it's updated every second and then the shelly is already rebooting.

I can't reproduce that bug, I'm able to revert my Plus 1PM back to stock. Im able to revert from 0.13 to hk to 0.12.

Could it be that you ever have flashed your shelly via a cable flasher or have you ever installed tasmota or esphome?

pascal9275, 2023-02-08:
Hello, actually no, it was completely stock and I wanted to test the HomeKit functions. As I am loosing the power measurement, I wanted to revert to stock.
After several tries, sometimes it goes up to 89% and sometimes 99%. In both scenarios I am not able to see the Shelly back in Shelly cloud or in 192.168.33.1 or the original IP. 
Using the command in this GitHub I am always able to revert back to HomeKit.

i am probably doing a wrong manipulation after the stock firmware is back? Can you let me know what action you take after the stock firmware is loaded?

timoschilling, 2023-02-08:
You need to reenable cloud before seeing there. On 192.168.33.1 you see them only in AccessPoint mode, but there you can’t do a revert. 

The simplest way to revert should be to open the web UI, take a look on the WiFi section memorize the ip of the Shelly, go down to the firmware section and klick the revert to stock button. Then after the progress stops wait some seconds and then type the ip into the address bar of your browser

pascal9275, 2023-02-09:
The way you described is exactly how I do.
I go to my browser and type the previously acquired IP
The system remains at this page
[image0.jpeg]

You mentioned I need to reenable cloud, how do I do that?

pascal9275, 2023-02-10:
I saw in the Shelly app, something called discovered device and I see my device, but for some reason it cannot connect to it.
[image]

megabyte0469, 2023-02-11:
Hello,
I have the same problem here with a Shelly plus 1PM. The uploading of the Homekit firmware was no problem. Unfortunately there is no way back. Firmware is loaded with 90%.
You come back to the Shelly interface but it always shows "Data is being loaded" more is not possible. 
I went back to the Homekit version and everything was back under Homekit. Can it be that you have to take out the Shelly first in Homekit completely before you go back to the original firmware?

pascal9275, 2023-02-11:
Finally, I managed to bring it back to stock firmware, here is how I proceed:
I removed the device from HomeKit and I selected a system reset.
The device goes then in AP mode.
I select the AP “wifi” and went to the address 192.168.33.1. 
the HomeKit information appeared back, I then re entered the wifi and gave the previously acquired IP.
On my browser I then typed the ip, the HomeKit page appeared and then I selected revert to stock.
On the Shelly app I went in discovered devices, and the system was able to add the device.

My understanding here is that there is still some parameters not resetting completely to stock, and this blocks the device on the revert to stock firmware action.

closed 2023-02-13 by timoschilling