mongoose-os-apps/shelly-homekit #1216: Can't set homekit pincode

roccodetoma, 2023-02-07:
Hi. I can't set homekit pin in my shelly2.5. It's not showing window. What's wrong? Thanx
[image]

sterl1i, 2023-02-07:
You have to click “Setup” ***@***.*** are such agreeable friends—they ask no questions, they pass no criticisms.

roccodetoma, 2023-02-07:
I did it but nothing changes. It keeps turning but no responses. I was waiting for a lot of minutes Rocco Detoma

timoschilling, 2023-02-08:
Sometimes it helps to make a device reset and then try it again.

roccodetoma, 2023-02-08:
I just tried to reset to factory default and install again but the problems isn't solved. Here's video. Any another solutions?

https://user-images.githubusercontent.com/124728625/217546488-e8936071-f5f8-4d28-8967-3f99ea2ff819.mov

timoschilling, 2023-02-08:
Please reset them again. Down in the systems section is a link to the log, please open them and then try it again. After some time copy the output from the log to this issue.

gkanb, 2023-02-14:
I have the same issue. I'm getting the error message: "Cannot read properties of undefined (reading 'replace')"

Here's the log:

> 388291894 shelly_debug.cpp:231    No log file, sending new entries
389036516 mg_rpc.c:314            Shelly.GetInfoExt via WS_in 192.168.33.2:61715
390036988 mg_rpc.c:314            Shelly.GetInfoExt via WS_in 192.168.33.2:61715
391037570 mg_rpc.c:314            Shelly.GetInfoExt via WS_in 192.168.33.2:61715
392035913 mg_rpc.c:314            Shelly.GetInfoExt via WS_in 192.168.33.2:61715
392356283 mg_rpc.c:314            Shelly.GetInfoExt via WS_in 192.168.178.24:54157
392923493 shelly_main.cpp:449     Up 392.92, HAP 0/0/16 ns 0, RAM: 206124/189012; st 120; disabled
393034222 mg_rpc.c:314            Shelly.GetInfoExt via WS_in 192.168.33.2:61715
393381096 mg_rpc.c:314            Shelly.GetInfoExt via WS_in 192.168.178.24:54157
394038776 mg_rpc.c:314            Shelly.GetInfoExt via WS_in 192.168.33.2:61715
394514908 mg_rpc.c:314            Shelly.GetInfoExt via WS_in 192.168.178.24:54157
395037904 mg_rpc.c:314            Shelly.GetInfoExt via WS_in 192.168.33.2:61715
395326138 mg_rpc.c:314            Shelly.GetInfoExt via WS_in 192.168.178.24:54157
396037507 mg_rpc.c:314            Shelly.GetInfoExt via WS_in 192.168.33.2:61715
397333573 W (397985) wifi:<ba-add>idx:3 (ifx:1, f6:20:1b:87:9f:94), tid:1, ssn:0, winSize:64
400923391 shelly_main.cpp:449     Up 400.91, HAP 0/0/16 ns 0, RAM: 207180/189012; st 120; disabled
408923336 shelly_main.cpp:449     Up 408.91, HAP 0/0/16 ns 0, RAM: 207180/189012; st 120; disabled
412427412 mg_rpc.c:314            Shelly.GetInfoExt via WS_in 192.168.178.24:54157
413355064 mg_rpc.c:314            Shelly.GetInfoExt via WS_in 192.168.178.24:54157
414380575 mg_rpc.c:314            Shelly.GetInfoExt via WS_in 192.168.178.24:54157
415398353 mg_rpc.c:314            Shelly.GetInfoExt via WS_in 192.168.178.24:54157
415831021 mg_rpc.c:314            HAP.Setup via WS_in 192.168.178.24:54157
416347925 mgos_sys_config.c:323   Saved to conf2.json
416372868 mg_rpc.c:314            Shelly.GetInfoExt via WS_in 192.168.178.24:54157
416924076 shelly_main.cpp:449     Up 416.91, HAP 0/0/16 ns 0, RAM: 207180/189012; st 120; disabled
424923318 shelly_main.cpp:449     Up 424.91, HAP 0/0/16 ns 0, RAM: 207180/189012; st 120; disabled

roccodetoma, 2023-02-14:
I just tried on another shelly 1 in another apartment with a different router. I managed to get the qr code for apple homekit. I completed the including in apple home kit as shown in video but when i press button in apple app nothing happen. The switch in the app is returning back without switching on the light

timoschilling, 2023-02-14:
@gkanb please open a separate issue and please ad a screenshot of that error message, I need some context here it appears. I think you have a different problem.

timoschilling, 2023-02-14:
@roccodetoma that looks like a different problem, please open a separate issue.

timoschilling, 2023-02-14:
I found the problem, the homekit pin is generated by the wifi name and that didn't work with names with a `€` in the name. I work on a fix.

timoschilling, 2023-02-14:
Could you test this fixed firmware build please. [shelly25-1216.zip](https://github.com/mongoose-os-apps/shelly-homekit/files/10737439/shelly25-1216.zip)
It works only for you Shelly 2.5, if you have other Shellys as well, let me know.

roccodetoma, 2023-02-19:
> Could you test this fixed firmware build please. [shelly25-1216.zip](https://github.com/mongoose-os-apps/shelly-homekit/files/10737439/shelly25-1216.zip) It works only for you Shelly 2.5, if you have other Shellys as well, let me know.

Thanks a lot Timoschilling....now it works perfectly....but i can't use it anymore with shelly, alexa and google home. Is it normal?

timoschilling, 2023-02-19:
Yes, we support only HomeKit and via web UI.

PR #1226 (2023-02-19): fix code gen with non alphanum chars ssid
closes #1216
closed 2023-02-19 by timoschilling