mongoose-os-apps/shelly-homekit #1217: Shelly not synchronizing

DiDeWi, 2023-02-09:
I installed 6 Shelly's (1L and 2.5). In most cases and after a while time in the shelly will synchronize. SNTP is google. But in some cases I can send commands to the Shelly 1L and that command will be executed (connection to WiFi OK) but I cannot set up weekly programming because time will not synchronize.
What do I miss?
Sincerely.
Dirk

timoschilling, 2023-02-09:
Sorry but I don't know what you are talking about? We do not support "weekly programming"

DiDeWi, 2023-02-09:
One of the settings of a Shelly device is its possibility to program fixed hours to put on and put off, called weekly programming. Shelly does not allow to program this setting when it is not synchronized with a SNTP. 
If I am on a wrong site, please concider this intervention as non existant.
Sincerely.
Dirk

timoschilling, 2023-02-09:
We do not offer that function of a weekly scheduling and don't plant to do that. We propose to do that via HomeKit.

Are you sure that you using our firmware or you are still on stock firmware. Which firmware version do you use?

closed 2023-03-11 by github-actions[bot]