mongoose-os-apps/shelly-homekit #1219: Method to control multiple outputs via API?

danricho, 2023-02-11:
First of all, thanks so much for this firmware! It's simply brilliant.

I have an RGBW2 in white only mode and I'd like to set 2 outputs to a different brightness in one API call to allow them to be controlled together?

I'm currently using the python requests library with the following GET requests...

```python
requests.get( url = 'http://shellyrgbw2-xxxxxx.local/rpc/Shelly.SetState', params = {
    "id": 3, 
    "type": 11, 
    "state": '{"brightness":50}'
})
requests.get( url = 'http://shellyrgbw2-xxxxxx.local/rpc/Shelly.SetState', params = {
    "id": 4, 
    "type": 11, 
    "state": '{"brightness":50}'
})
```

The problem is that the first output fades before the second output command is processed.

I also tried a version which performs the two GET requests in separate threads so that the response receipt doesn't block the second request, however I think the Shelly device is blocking the second response on it's side...

I was hoping for something with a single call so that they fade together:
```python
paramArray = [
    {
        "id": 3, 
        "type": 11, 
        "state": '{"brightness":50}'
    },  {
        "id": 4, 
        "type": 11, 
        "state": '{"brightness":50}'
    }
]
requests.get( url = 'http://shellyrgbw2-xxxxxx.local/rpc/Shelly.SetState', params = paramArray )
```

Is there a method to do this?

Thanks again!

timoschilling, 2023-02-13:
In my tests there was a delay between the first and the second one of ~100ms (I used curl to send two commands in series). Then I tried to make test implementation of your request and end up by ~90ms, so I think it doesn't make sense to implement this. Sorry.

danricho, 2023-02-13:
@timoschilling Does the fading occur concurrently in your tests?

Is there any chance you can share your method? I admit I didn't try with curl so I will do that as well.

Cheers!

danricho, 2023-02-16:
For anyone else who may have this type of issue, I tracked the issue down to the resolution time to translate the host name of the Shelly device to its IP address. In other words, use the ip in the urls to make everything near instantaneous.

Cheers!