mongoose-os-apps/shelly-homekit #1220: Shelly RGBW Blinking and Restarting

90III01, 2023-02-11:
I just installed 3 Shelly RGBW2 with a led RGBW stripe. Same issue on all: with original firmware it works fine, after Homekit Update it start blinking (1 long - pause - 6 fast) both shelly and stripe (in red color) and is impossible to connect with wifi. AP mode works for a few seconds (Homekit is installed so OTA flash was done) and then continue to disconnect. It is not discoverable at all in my wifi network. Any solution ?

90III01, 2023-02-14:
I found the origin of the problem but not the solution. Shelly read the AC signal as 5 consecutive input, so reset and go in AP mode. If i put the switch after transformer or detach it the Shelly works perfectly. It is obviously a software problem only in HomeKit firmware. So please urgently fix this firmware bug.

timoschilling, 2023-02-14:
At the moment we didn't support the AC converter. If you cloud don't use the AC converter and switch your shelly with 12/24V. If not, there are possibilities to solve that problem by wiring a capacitor to the input of the shelly. If that could be a solution for you, please google, I didn't know any details.

closed 2023-02-14 by timoschilling

90III01, 2023-02-15:
Thank you for the reply. I don’t understand the problem: I’ve several Shelly1 in my home and all works perfectly with AC. Just copy the same software behaviour (I suppose ignoring state changes for some sec after first signal) and the problem is solved. I’m missing some point?

timoschilling, 2023-02-15:
You can't compare that both models. The 1 is direct wired to AC and had a good input, for example the 1L and the I3 aren't good. The RGBW2 has an input which is build for DC and not for AC. That's why you need to use the this AC adapter. The problem is that this adapter produces noice, which means the signal isn't stable on/off, it's a mostly on instead of a pure on. That triggers the reset.

In theory we could use the stabilisation from the 1L / I3, but in practice not, because it can't run at the same time with the dimming function of the RGBW2. Both functions needs a hardware timer, but the cpu has only one. There is a big refactoring needed which make it possible to share that timer. Depending on the complexity of that problem and the time of the developers, it's not clear when that will happen, possible never.

I only can suggest to switch direkt via 12/24V or to stabilise the adapter with a capacitor.

90III01, 2023-02-15:
Thank you again for the reply. Now I understand the problem, a mix of hardware and software limitations. As I need to keep my actual physical switches setup and is not sure if and when the fix could be implemented I definitely need to “stabilise” the signal from the 220 AC to a DC 24V. I’ll work on it. Thank you for clarifications.