mongoose-os-apps/shelly-homekit #1221: shelly 1 stuck on loading after trying to flash homekit

Mymi42, 2023-02-12:
Hello everyone,

After using the new link (for all shelly devises) to flah my shelly 1 for homekit, I have the same message for one hour trying to access its IP address 192.168.X.X. "Shelly homekit - loading"
I guess something goes wrong and now I cannot reach the shelly through the shelly cloud app anymore... 

Do you have any idea to help me ??

Best regards,

douweh, 2023-02-12:
Hi, I had the same issue...
Then found out I just did a refresh in my browser and so was accessing the shelly still through the update link (ie `http://A.B.C.D/ota?url=http://shelly.rojer.cloud/update`)..

When I tried _just_ the ip adress (like a minute after the update initiating) all worked.

Mymi42, 2023-02-12:
Thank you for your answer but it did not seems to work for me. I stil have the same message...
If you have an other idea, it will be my pleasure to try :-)

regards.

timoschilling, 2023-02-21:
@Mymi42 Could you please tel me the exact IP address. It's normal that the shelly isn't available to the shelly app anymore, we only support HomeKit not the app.

Mymi42, 2023-02-21:
192.168.1.62
I know I won’t find the shelly on shelly cloud anymore after the flash but I should have a page to configure homekit and I don’t. The ip address only display « loading… » with HomeKit logo.

timoschilling, 2023-02-21:
Which browser do you use?

Mymi42, 2023-02-21:
Safari

timoschilling, 2023-02-21:
Please do the following things:
1. open settings of safari `cmd + ,`
2. open "Advanced" tap
3. activate "Show Develop menu bar"
4. close settings panel
5. press `command + option + i` 
6. click on "Console"
7. Open http://192.168.1.62/debug/log?follow=1 and keep it open
8. open the web ui in a new tap
9. copy the content of the debug log tap to this issue

Mymi42, 2023-02-21:
Ok, I will do this tomorrow. I don’t have my computer and the shelly is unpluged right now. 
thanks for your help.

Mymi42, 2023-02-22:
Hello :
213306322 shelly_debug.cpp:231    No log file, sending new entries
213581114 mgos_mongoose.c:66      New heap free LWM: 27696
216617326 shelly_main.cpp:443     Up 216.61, HAP 0/0/12 ns 0, RAM: 30788/27696; st 0; 0.1: st:0 in_st:0 inm:1 ininv:0
224617357 shelly_main.cpp:443     Up 224.61, HAP 0/0/12 ns 0, RAM: 30788/27696; st 0; 0.1: st:0 in_st:0 inm:1 ininv:0
232617262 shelly_main.cpp:443     Up 232.61, HAP 0/0/12 ns 0, RAM: 30788/27696; st 0; 0.1: st:0 in_st:0 inm:1 ininv:0
240617147 shelly_main.cpp:443     Up 240.61, HAP 0/0/12 ns 0, RAM: 30788/27696; st 0; 0.1: st:0 in_st:0 inm:1 ininv:0
248617335 shelly_main.cpp:443     Up 248.61, HAP 0/0/12 ns 0, RAM: 30788/27696; st 0; 0.1: st:0 in_st:0 inm:1 ininv:0
256617334 shelly_main.cpp:443     Up 256.61, HAP 0/0/12 ns 0, RAM: 30788/27696; st 0; 0.1: st:0 in_st:0 inm:1 ininv:0
hope I did your procedure well. I was not sure if I needed to tap http://..... on the tab or in the console window...
Does it help you to understand ?

Hrn737, 2023-03-02:
> Safari
I had the same problem, switch to another browser, I did the same on Chrome and open instantly

Mymi42, 2023-03-04:
Woauuuh ! I have tried with Firefox and it works too. Imbelivable… Never thought it could came from the browser…
Thank you very much. 
If someone can mention it somewhere in the procedure, it may help others. 
thank you once again!

Hrn737, 2023-03-04:
Glad to help!

timoschilling, 2023-03-04:
@Hrn737 thats for the tip

closed 2023-03-04 by timoschilling