mongoose-os-apps/shelly-homekit #1222: Doorbell not supported

gnoto, 2023-02-12:
I configured a Shelly 1 Plus as a Doorbell. Everything seems correct but the Home App shows the accessory as not supported and does not allow any configuration.
Any idea?
[image]
[image]
[image]

timoschilling, 2023-02-13:
That's a common issue, the Home app didn't support it / has no UI for it. But it will work anyway, you will receive notifications if someone rings the bell.

closed 2023-02-13 by timoschilling

insidegui, 2023-04-11:
Is there any special setup required in order to get notified when the doorbell is activated? I see it as unsupported in the Home app as well, pressing the switch makes the doorbell ring sound in all HomePods, but I'm not being notified on any device.

ghaithster, 2023-07-09:
Hoping to get assistance… my building has a four wire intercom set up (12vdc) 
[image]
with basic panels in each apartment (Listen, Talk, Door). I successfully hooked up the Shelly 1 and slashed it with the HomeKit software. However, I can’t figure out how to get notified when someone presses my apartment button outside. My apartment panel does ring and the wire marked X (see photo) has the tone signal. I connected this wire to the SW of the Shelly 1, and I think I can figure the input correctly, as detached, and then doorbell. I’m pretty sure I’m missing something…

nachomezzadra, 2024-12-04:
> Is there any special setup required in order to get notified when the doorbell is activated? I see it as unsupported in the Home app as well, pressing the switch makes the doorbell ring sound in all HomePods, but I'm not being notified on any device.

i'm having this exact same issue.  It was working with previous fw/ios versions, but now i'm not getting any notifications at all.  iOS v18.1.1, fw version 2.13.3 (build 20241203-191940/2.13.3-g1cb5084).  Has anyone been able to sort this out?