mongoose-os-apps/shelly-homekit #1223: my garage door opens itself!

louisdcmr, 2023-02-13:
Hello, I come to you because I have a problem.
My garage door is equipped shelly 1 flashed and in garage door mode.
and everything worked well for 6/7 months or more but the ca for 2/3 weeks that my garage door opens alone and that I receive a notification on homekit out of person did not trigger the opening and I would like to know if you think it comes from the garage door or if Is that the shelly setting itself? I’m worried about security concerns, especially since it never happened to me.

thanks in advance.

timoschilling, 2023-02-13:
Since this would be the first time in 2 1/2 years (I mean the time that garage door mode exists), I think it's a problem with your specific setup.

louisdcmr, 2023-02-13:
and what would you recommend?

closed 2023-02-21 by timoschilling