mongoose-os-apps/shelly-homekit #1225: Can't generate QR-Code for HomeKit with Shelly Plus 1

gkanb, 2023-02-14:
Hello,

after flashing the HomeKit Firmware for my Shelly Plus 1. I can't generate a QR Code. After clicking "Setup" in the HomeKit Settings, I'm getting the error message: 

> Cannot read properties of undefined (reading 'replace')

[image]

The Debug log is as following:

```
388291894 shelly_debug.cpp:231 No log file, sending new entries
389036516 mg_rpc.c:314 Shelly.GetInfoExt via WS_in 192.168.33.2:61715
390036988 mg_rpc.c:314 Shelly.GetInfoExt via WS_in 192.168.33.2:61715
391037570 mg_rpc.c:314 Shelly.GetInfoExt via WS_in 192.168.33.2:61715
392035913 mg_rpc.c:314 Shelly.GetInfoExt via WS_in 192.168.33.2:61715
392356283 mg_rpc.c:314 Shelly.GetInfoExt via WS_in 192.168.178.24:54157
392923493 shelly_main.cpp:449 Up 392.92, HAP 0/0/16 ns 0, RAM: 206124/189012; st 120; disabled
393034222 mg_rpc.c:314 Shelly.GetInfoExt via WS_in 192.168.33.2:61715
393381096 mg_rpc.c:314 Shelly.GetInfoExt via WS_in 192.168.178.24:54157
394038776 mg_rpc.c:314 Shelly.GetInfoExt via WS_in 192.168.33.2:61715
394514908 mg_rpc.c:314 Shelly.GetInfoExt via WS_in 192.168.178.24:54157
395037904 mg_rpc.c:314 Shelly.GetInfoExt via WS_in 192.168.33.2:61715
395326138 mg_rpc.c:314 Shelly.GetInfoExt via WS_in 192.168.178.24:54157
396037507 mg_rpc.c:314 Shelly.GetInfoExt via WS_in 192.168.33.2:61715
397333573 W (397985) wifi:idx:3 (ifx:1, f6:20:1b:87:9f:94), tid:1, ssn:0, winSize:64
400923391 shelly_main.cpp:449 Up 400.91, HAP 0/0/16 ns 0, RAM: 207180/189012; st 120; disabled
408923336 shelly_main.cpp:449 Up 408.91, HAP 0/0/16 ns 0, RAM: 207180/189012; st 120; disabled
412427412 mg_rpc.c:314 Shelly.GetInfoExt via WS_in 192.168.178.24:54157
413355064 mg_rpc.c:314 Shelly.GetInfoExt via WS_in 192.168.178.24:54157
414380575 mg_rpc.c:314 Shelly.GetInfoExt via WS_in 192.168.178.24:54157
415398353 mg_rpc.c:314 Shelly.GetInfoExt via WS_in 192.168.178.24:54157
415831021 mg_rpc.c:314 HAP.Setup via WS_in 192.168.178.24:54157
416347925 mgos_sys_config.c:323 Saved to conf2.json
416372868 mg_rpc.c:314 Shelly.GetInfoExt via WS_in 192.168.178.24:54157
416924076 shelly_main.cpp:449 Up 416.91, HAP 0/0/16 ns 0, RAM: 207180/189012; st 120; disabled
424923318 shelly_main.cpp:449 Up 424.91, HAP 0/0/16 ns 0, RAM: 207180/189012; st 120; disabled
```

timoschilling, 2023-02-14:
Which browser do you use?

gkanb, 2023-02-14:
I've tried out with Chrome. With Firefox I'm getting the message:

> info.code is undefined

timoschilling, 2023-02-14:
What's the name of your wifi?

gkanb, 2023-02-14:
It contains an `&` and `-` symbol. I could try to use an WIFI name without a special character but alphabetical only tomorrow morning.

timoschilling, 2023-02-14:
here is a fixed build, it only works for the Plus1, if you have any different shelly types let me know.
[shelly-plus1-1216.zip](https://github.com/mongoose-os-apps/shelly-homekit/files/10737594/shelly-plus1-1216.zip)

First you issue looks different, but now it looks the same as #1216, sorry for that. I will close this issue now as double. Please report feedback here: #1216

gkanb, 2023-02-15:
Thanks for the build. However it seems like the issue still persist. I'm getting the same error message. I've tried to connect to a wifi called "Test", but it still failed to generate an QR code. Maybe this is not about the wifi name?

roccodetoma, 2023-02-15:
> 

Thanks but how do i install this build from the file you attached? I was installing the previous ota. I checked the original ui but i didn't find the way to upload from my mac

timoschilling, 2023-02-15:
If you are on original firmware, you need first do a ota to homekit firmware and then you can upload it via the web ui.
@roccodetoma please take attention, the file in this issue is for the Shelly Plus1, in the other [issue](https://github.com/mongoose-os-apps/shelly-homekit/issues/1216) is one for the Shelly 2.5. You need to take the right one for your Shelly.

closed 2023-03-17 by github-actions[bot]