mongoose-os-apps/shelly-homekit #1229: Shelly 1v3 Doorbell

nullatore, 2023-02-17:
I configured Shelly 1v3 as Doorbell (latest firmware version) but I can't figure out how to configure correctly input Input Mode in order to have reliable notifications from Apple Homekit.

If I set mode "Momentary" system raises notification only on "last event" status change ,e.g. if last event is "single press" I have to make a "long press" (means long signal on SW port) to obtain notification and viceversa.

Mode "Toggle, on=off = single press" seems not working properly.

Mode "Toggle on= single , off = double" seems to be my best choise but somethings it doesn't work. I don't still figure out the logic behind and so to recognize why.

Any idea?
Thanks.

timoschilling, 2023-02-17:
I would say momentary is the right setting. What you describe sounds like a bug, I will take a look on that.

nullatore, 2023-02-23:
> I would say momentary is the right setting. What you describe sounds like a bug, I will take a look on that.

Did you have any chance to take a look at? Tnx

nullatore, 2023-03-07:
> What kind of switch do you use?

I use Lock type and Switch type, both of them with Detached input (= Doorbell)
For both I have that issue.

[image]
[image]

timoschilling, 2023-03-15:
"confirmed-bug" is only premature until I have found time to test it

snaggedprisma, 2023-06-14:
are there any news? i tryed to set it up but dosent work :( i have a shelly uni and put it to input when someone press on the doorbell button then is the signal comming to that input. when i put the shelly to homekit he regonize it as a doorbell and ask for picture of face and if i want to have ah notification on homepod. but after it dosent work and i cant change something in the settings to bring a notification :( please help

morettoni, 2025-03-28:
No updates since 2023... any news and/or test? I was thinking to use a spare Shelly1 to detect if someone rings and the secondary door...

nullatore, 2025-03-28:
@morettoni 
have you considered on changing FW?

morettoni, 2025-04-03:
> [@morettoni](https://github.com/morettoni) have you considered on changing FW?

yes, but other FW are not so user friendly for the configuration, and now that I am getting older I prefer a simple way to manage devices at home... for example I don't like to play with "complex" JSON files...

rinoronie, 2025-10-27:
news?