mongoose-os-apps/shelly-homekit #1231: The problem is not solved. I applied as a solution back to the previous version(Firmware Version:2.11.1). With this version it works correctly. Please check this latest version, as it is not working properly. Thanks.

marmitol, 2023-02-18:
The problem is not solved. I applied as a solution back to the previous version(Firmware Version:2.11.1). With this version it works correctly. Please check this latest version, as it is not working properly. Thanks.

_Originally posted by @marmitol in https://github.com/mongoose-os-apps/shelly-homekit/issues/1157#issuecomment-1435788320_

marmitol, 2023-02-18:
I recently outfitted a friends house with 10 Shelly 2.5 PM for roller shutter control.
Two of them are currently showing the following behavior.
They

are displayed as "closed" in Homekit
are really "open"
don't react to the physical buttons
No action is possible anymore, neither via Homekit, nor via the wall buttons.

The only recovery I found so far is to start a new calibration, but for the ones that show this behavior it returns latest after a few days.

I have yet to look at the logs, because I only sporadically have access to the devices, but will add them here once I got them.

timoschilling, 2023-02-21:
double of #1186

closed 2023-02-21 by timoschilling