mongoose-os-apps/shelly-homekit #1233: Bridge support for battery based devices

timoschilling, 2023-02-21:
HomeKit over IP isn't requires a consistent IP connection (WiFi / Ethernet) to work properly, only HomeKit over Bluetooth is made for battery bases devices. To support battery based Shelly's we plan to add a bridge mode to support them.

This issue replaces #104, #124 and #295.

vrpt2022, 2023-02-22:
@timoschilling Thanks for this!!!

Do you think #1168 could be included as well here?

timoschilling, 2023-02-22:
@vrpt2022 yes could be

vrpt2022, 2023-03-23:
Good morning!

Any update regarding this?

thanks

timoschilling, 2023-03-23:
We don’t have any route map and no ETAs. The enhancement label only means that we think it’s a good idea and that it maybe some day will be implemented. The only thing that I can say that I’m also interested in this feature.

caarlos0, 2023-10-22:
FWIW I have something working for shelly flood here: https://github.com/caarlos0/homekit-shelly