mongoose-os-apps/shelly-homekit #1234: Shelly1v3 + Carplay + Siri

nullatore, 2023-02-25:
Scenario:
Shelly1v3 + Carplay

When iphone's working with Carplay and is locked , Siri doesn't allow to execute any command toward my Shelly accessory (it says to unlock the phone before).
At the same condition (carplay+phone locked) no problem with other HomeKit certified accessory.

Everything's fine with my shelly if iphone is unlock.

timoschilling, 2023-02-25:
What kind of accessory is your Shelly, a garage door or a switch?

nullatore, 2023-02-25:
> What kind of accessory is your Shelly, a garage door or a switch?

A switch type lock , input mode = toggle

[image]

vrpt2022, 2023-02-25:
> > What kind of accessory is your Shelly, a garage door or a switch?
> 
> A switch type lock , input mode = toggle
> 
> [image]

As far as I know SIRI don’t allow security accessories to be activated with a locked phone. Locks/Doors can’t be open through Siri if you don’t unlock the phone. If you have an Apple Watch or HomePod you can try to ask to open the garage door and it will reply that it must be done through unlocked iPhone. Here some comments:
https://www.reddit.com/r/HomeKit/comments/tk8j1s/cant_ask_homepod_to_open_my_garage_via_ismartgate/

nullatore, 2023-02-25:
> As far as I know SIRI don’t allow security accessories to be activated with a locked phone. Locks/Doors can’t be open through Siri if you don’t unlock the phone. If you have an Apple Watch or [...]

I think it isn't my case.
With my my iphone locked, I successfully can launch command to all my accessory (included Shelly) from AppleWatch and Ipad.

Furthermore, Siri says me that I have to unlock iphone (not to continue this operation on that device).

timoschilling, 2023-02-25:
> If you have an Apple Watch or HomePod you can try to ask to open the garage door and it will reply that it must be done through unlocked iPhone.

That’s not right. If your Apple Watch isn’t locked you can open secured devices. 

But in fact your answered your initial issue by your self. Secured devices (lock, garage door, etc.) can’t be open via Siri if a device is locked.
You can only control devices like light bulbs etc if locked. 

In general it’s not up to use how siri works with a Shelly. Because siri only controls HomeKit. The Shelly didn’t have any knowledge about siri.

closed 2023-02-25 by timoschilling

nullatore, 2023-02-25:
> That’s not right. If your Apple Watch isn’t locked you can open secured devices.
>
> But in fact your answered your initial issue by your self. Secured devices (lock, garage door, etc.) can’t be open [...]

My Shelly configured as a light bulb has the same problem.
My Smart Plug (certified accessory) hasn’t the problem and work also with locked iPhone.
Hmmm…