mongoose-os-apps/shelly-homekit #1236: Shelly uni input unstable

Chancerelle, 2023-02-28:
Hello
I have connected a Shelly uni to control my garden gate.
It is connected to a 24vAC power supply.
I could implement it without any issue to homebridge with the proposed firmware. I am using only switch and detached input.
My issue : when I open the garden gate, the input information (linked to a 0 - 24AC plug on the controler of the gate) is really unstable : it change more than every 0,5s from close to open if I look to the web interface and randomly every 5 or 10s in homekit application.
Is there any possibility to "filter" the input information so that it is consistent with the input signal of the gate ?
In addition, the input information is fully stable using the original firmware from shelly.
thank you for your help

timoschilling, 2023-02-28:
Please flash your Shelly UNI with this firmware and test it.
[shelly-homekit-ShellyUNI.zip](https://github.com/mongoose-os-apps/shelly-homekit/files/10849838/shelly-homekit-ShellyUNI.zip)

Chancerelle, 2023-02-28:
It is great !
Seams to solve my issue. I will chek again latter, but first feedback is good.
Could you please tell me what is this firmware ? What was modified ?

timoschilling, 2023-02-28:
Your issue is a double of #715 and was fixed in https://github.com/mongoose-os-apps/shelly-homekit/pull/1160, but not yet released. That build contains all changes since the last release.

closed 2023-02-28 by timoschilling