mongoose-os-apps/shelly-homekit #1237: RGBW2 Fails to flash

Colins138, 2023-03-01:
Hi I have two RGBW2’s which failed to flash using the same process.  I connected the device using the Shelly app and then the firmware was updated to the latest version using the Shelly web interface.

Using the general script pointing to the IP (in my case specifically “ http://192.168.1.149/ota?url=http://shelly.rojer.cloud/update”) I started the update.

The immediate message returned was;

{"status":"updating","has_update":false,"new_version":"20221027-095909/v1.12.1-ga9117d3","old_version":"20221027-095909/v1.12.1-ga9117d3"}

The lights on the RGBW2 have been flashing morse code now for 30 minutes, and I have now tried to reach it with no success.

timoschilling, 2023-03-01:
Are you connected to the wifi of the shelly when trying to reach 192.168.33.1 or to your home wifi?

Colins138, 2023-03-01:
I try to connect directly to the Shelly Wi-Fi but it keeps dropping out and never quite able to properly attach to the network.  It’s as if the Shelly is powering itself up and down as the lights it is connected to flash.

Colins138, 2023-03-01:
I’m just attaching a short video so that you can see what the device is doing.  The lights it is attached to flash with the lights on the RGBW2.

As I said, this is the second device this has happened to now so it seems to be a consistent issue.

Perhaps I’m misreading the message when I tried to flash the device but it looks as though it wasn’t happy with the Shelly firmware.

I’m just a User not a coder so forgive the ignorance.

timoschilling, 2023-03-02:
I think you missed to add the video.

Colins138, 2023-03-02:
https://user-images.githubusercontent.com/126680877/222406288-de7762d4-62bb-4d3a-acc7-6095f8dd44ed.MOV

Colins138, 2023-03-02:
I'm not sure why it didn't upload with the previous message but hopefully you can now see this.  The lights in the room below flash to the same cadence as the light on the RGBW2.

When I flashed the first RGBW2 the lights were on in the Shelly app.  When I flashed the second, they were off.  They started this flashing (red channel only, if that makes a difference) as soon as the mongoose firmware was applied.

I've flashed a number of Shelly 1L's successfully using exactly the same process so this wasn't my first use either.

timoschilling, 2023-03-02:
You just posted a picture and not a video.

In general it's the same to flash the 1L or the RGBW2.

Colins138, 2023-03-02:
Yes for some reason it shows as a video for me but here’s an Imgur link.  I don’t know if it adds a lot but if it helps diagnose issues then so much the better.

https://imgur.com/gallery/FEekPEf

timoschilling, 2023-03-02:
LOL no I see the video in the comment from before. 

For me it looks like that maybe the power supply didn’t deliver a stable voltage. Please try a different one and without leds wired.

Colins138, 2023-03-11:
Hi Timo.  I have been away so couldn't do more.  I don't have a separate power supply to use, but disconnected the LED's so there's no power draw and it's still doing the same thing, and won't create a stable Wi-Fi signal.

Both of the RGBW2's were connected to the same supply and both worked perfectly with the Shelly app.  They also both updated Shelly firmware without issue.  It was only installing this firmware when it started to fault.

Obviously don't know if others have experienced the same issue but seems odd for two to go.  I'm reluctant to get another RGBW2 but don't know if there are alternatives.

timoschilling, 2023-03-11:
Are you using the Shelly ac/dc converter on the input? Happens your problem if you disconnect the wire on the input?

QBJack, 2023-03-22:
Hi guys,

I have a similar problem to flash my RGBW2. It's WIFI is stable and an update to the latest shelly firmware works without issues. It's just not able to flash with the firmware provided here. 

As said, WIFI and power supply are stable.

Greets
Thomas

QBJack, 2023-03-22:
Just after posting my first comment I did a reload in my browser and there it is - the flash worked. I can't really tell what fixed this, maybe an update to a newer shelly firmware or maybe because I have waited longer before reloading.

So @Colins138 give this a try and wait longer and/or do a default shelly firmware update._

Colins138, 2023-04-18:
Just leaving a final comment before the item goes stale because it seems to me there is an issue with this firmware and it might be useful for others to know the circumstances.  The Shelly works with its native firmware and simply doesn't with the HomeKit version. The circumstance are that it is connected to a Meanwell 240w LED driver and a total of 16m RGB led strip (two separate runs) at 15w per metre which comes to exactly 240w.  In practice it was cut down a little to make this very slightly under.

Perhaps there's a headroom issue, but if so this should serve as a useful warning to others that the HomeKit firmware is sensitive to this.

closed 2023-06-23 by github-actions[bot]

AustinChopra, 2024-03-28:
i was able to get it to flash after a lot of trial and error, what worked for me are as follows,

1. Register device with shelly app and find ip address
2. download the shelly homkit tools
3. download the shellyrbg2 fimware
4. put the firmware in the tools folder with the flash-shelly.py file
5. run .\flash-shelly.py --local-file=shelly-homekit-ShellyRGBW2.zip <ip address> in cmd, you might need to repeat this step multiple times to untill your successful
6. run .\flash-shelly.py <ip address> in cmd

For some reason the RGBW2 reports a firmware version on 0.0.0 initially, after running step 5 it will flash it with firmware v1.14.0 and then you can update it to the latest homekit firmware. Also when i ran step 5, the script froze on me after flashing the shelly firm ware, i had to close cmd and reopen it and then i was able to run step 6

Austinthemighty, 2024-03-28:
i was able to get it to flash after a lot of trial and error, what worked for me are as follows,

1. Register device with shelly app and find ip address
2. download the shelly homkit tools
3. download the shellyrbg2 fimware
4. put the firmware in the tools folder with the flash-shelly.py file
5. run .\flash-shelly.py --local-file=shelly-homekit-ShellyRGBW2.zip <ip address> in cmd, you might need to repeat this step multiple times to untill your successful
6. run .\flash-shelly.py <ip address> in cmd

For some reason the RGBW2 reports a firmware version on 0.0.0 initially, after running step 5 it will flash it with firmware v1.14.0 and then you can update it to the latest homekit firmware. Also when i ran step 5, the script froze on me after flashing the shelly firm ware, i had to close cmd and reopen it and then i was able to run step 6