mongoose-os-apps/shelly-homekit #1238: Shelly Plus 1PM randomly turns off

OleHornsby, 2023-03-03:
First of all, thank you for your great work. I have been using a Shelly Plus 1PM in a two light switch configuration since September of 2022, with the Homekit software without any issue. It even worked better than the philips hue lights.

But this last monday, it all of a sudden would shut off completely, turn off the light in the process and be unresponsive to pressing the physical light switches or to any commands via Homekit or Web interface.

I tried getting a Log but the moment it turns off, the Log stops and then I can only get one after it rebooted.

My setup is 2 HomePod Minis as Home Hub and an Ubiquiti Amplifi Mesh Network but neither of those changed in the last 2 months.

I now have the standard Shelly software on it again and it has absolutely no issue.

Could the Shelly itself be broken and therefore have an issue with the Homekit software?

I don't believe it has to do with newest Homekit Updates as I did a complete factory reset, flashed the homekit software and didn't connect it to my home at all and it still rebooted in random intervals.

Thank you for your help and I'll gladly provide more info if necessary.

timoschilling, 2023-03-03:
You can enable file logging with `http://a.b.c.d/rpc/Config.set?reboot=true&save=true&value={"file_logger":{"enable":true}}`
and disable with `http://a.b.c.d/rpc/Config.set?reboot=true&save=true&value={"file_logger":{"enable":true}}`.
Please don't forget to disable otherwise you will damage the flash of your shelly.

nliviu, 2023-03-03:
The second command (to disable logging) should have `"file_logger":{"enable":false}`

OleHornsby, 2023-03-03:
Thank you for the quick reply. So I have two logs, one was created shortly after a restart and was going during one or up to one restart and the other log was created directly after a restart, when I was able to click on the "log" button again.

When I say I have two switches, in german it's called a "Wechselschaltung" I think it's a Changeover Switch in english?
Two switches on opposite sides of the room control one light source. So the shelly is configured as a Switch with "Edge" setting so that both light switches work and I can control the shelly via Homekit.

What I already saw but don't really know what it means: In the log "during crash" it says WiFI STA Disconnect Reason: 8

Attached are the logs, if you need them as clear text, I also have them as a simple .txt file
[Shelly_Log_After_Crash.pdf](https://github.com/mongoose-os-apps/shelly-homekit/files/10885572/Shelly_Log_After_Crash.pdf)
[Shelly_Log_During_Crash.pdf](https://github.com/mongoose-os-apps/shelly-homekit/files/10885573/Shelly_Log_During_Crash.pdf)

timoschilling, 2023-03-07:
`Shelly_Log_During_Crash.pdf` doesn't look like after a crash, it more looks like after a reboot because of a config change. Please sample a new log.

rs38, 2023-03-10:
I am currently also investigating in strange turn-offs of my 1PM since a week or so.
Currently I have no clue why the switch is turning off after ~15 min...

As Apple HK apparently does not provide any logging or tracing i'd like to try with the file logging meantioned above. how long can it run without "damaging the flash"?

timoschilling, 2023-03-10:
@rs38 If you say yours is rebooting and not turn off like in this issue. Please open a new issue. 

Sort answers for your questions: it would take multiple months to kill the flash with turned on logs. Multiple of my Shelly’s had it on for 1/2 year and nothing happened. If the reboot happens more and more early, it could be a hardware issue. 

If you like to write more please open a new issue.

rs38, 2023-03-11:
it *was* turning off until I changed the initial state of the shelly to "on"
I can open a new issue,

OleHornsby, 2023-03-20:
@timoschilling I'm sorry for the late reply, had a lot going on in the last week.
Is the Shelly storing the log files somewhere and is there a way for me to download them? 
As of right now, I'm connecting to my Shelly with the Homekit Firmware and under "System" I click on "Log", which opens a seperate page which is updating continously.
Now comes my issue: When my Shelly Reboots / Turns off / gets unresponsive, the Log file isn't updating anymore and eventually gives me a "Page loading" error.
If I then click on the "Log" button again after the Shelly is up again, it starts a completely new log.

I put in the http://a.b.c.d/rpc/Config.set?reboot=true&save=true&value={"file_logger":{"enable":true}} command and it's giving me a confirmation message.

So as of right now, I can get a log up to the shutdown, and afterwards, but no Info during.

One thing I also noticed: You mentioned that turning on the file logger can damage the Flash. When the problem started, I could look at the log file and it looked the same as it does now, but I never activated the file_logger via the command you gave me.

Could it be that, by some weird bug, file logging was turned on all the time since I got the Shelly, and now the flash drive is damaged and causes this issue?

jeffnhelen, 2023-03-25:
[Shelly 1+PM Persistant Log.pdf](https://github.com/mongoose-os-apps/shelly-homekit/files/11069420/Shelly.1%2BPM.Persistant.Log.pdf)

I recently updated from 2.11.0-alpha8 as I would periodically (every few days...sometimes longer) have no response issues with HK.  After upgrading to 2.11.2 Master I see even more frequent issues (every few hours...sometimes minutes).  I have enabled the log save option and was able to upload a log file to reference as well.  I would notice that logging would stop refreshing while watching it so it seemed that's when a reboot happened and I would have to start a new log window as soon as I could get back into the main page.

It appears that the last messages show either HomeKit communication or me looking at the webpage (watching the uptime), but then the counter resets and first message is RPC/UART0.  I was able to capture this same event during two crashes:

132053768 mg_rpc.c:314            Shelly.GetInfoExt via WS_in x.x.x.145:64842
139578686 shelly_main.cpp:449     Up 139.57, HAP 0/1/16 ns 1, RAM: 208320/177276; st 40; 0.1: st:0 in_st:0 inm:3 ininv:0; 6.1: svt:6 inv:0 
141384737 mgos_http_server.c:180  0x3ffc7be0 HTTP connection from x.x.x.145:64846
3200073 mgos_rpc_channel_ua:320 0x3ffc79e0 UART0
3262707 shelly_reset.cpp:222    Reboot counter 2
3355345 mgos_sys_config.c:323   Saved to conf9.json
3375599 shelly_main.cpp:796     === Creating peripherals
3384179 shelly_output.cpp:43    OutputPin 1: pin 26, on_value 1, state off
3400856 shelly_input_pin.cpp:44 InputPin 1: pin 4, on_value 1, state off
3406456 shelly_pm_bl0937.cpp:71 BL0937 @ 5/18/23 apc 1.867207
3411753 shelly_noisy_input_:107 Starting sampling timer
3469881 shelly_noisy_input_:118 NoisyInputPin 0: pin 25, on_value 0, state off mc 1 0x2000000
3475303 shelly_main.cpp:255     === Creating accessories
3481875 shelly_switch.cpp:245   Exporting 'Shelly SW': type 0, state: 0
3493188 shelly_main.cpp:304     === Starting HAP bridge (ID XX:XX:XX:XX:XX:XX, UUID <n/a>, CN 11, 2 accessories)
3513500 HAPPlatformTCPStrea:187 Listening on 9000
3523603 mgos_init.c:36          Init done, RAM: 273612 total, 232776 free, 232776 min free
3529797 shelly_main.cpp:341     HAP server state: 1

Not sure if it's unrelated, but most times after reboot the Wifi is disabled and I have to access it via the AP to re-enable Wifi and get it connected back to my network.

closed 2023-04-23 by github-actions[bot]