mongoose-os-apps/shelly-homekit #1239: Shelly 1L - not switching state

kingwalius, 2023-03-05:
Hi together, 
I've installed the firmware on the Shelly 1L and after some time the Shelly did not respond to the action made within Homekit. 
The Shelly was still active within Homekit. But the relay was not responding to the action. 

I tried to reinstall the Shelly. This worked sometimes but not every time. 

Do you guys have any suggestion? 
Thanks very much in advance. Looking forward to your response. 

Best regards 
Mario

timoschilling, 2023-03-07:
Can you please open the log of your shelly, link is in the system section. Then use it in HomeKit and past the log output from that time here.

kingwalius, 2023-03-29:
Currently the issue is not happening at the moment, of it happens again I will give feedback immediately.

Best regards
Mario

closed 2023-03-29 by timoschilling

kingwalius, 2023-04-01:
Hi together,
apparently the issue came back just yesterday.
Attached you‘ll find a screenshot of the logfile.

Best regards
Mario

Closed #1239<https://github.com/mongoose-os-apps/shelly-homekit/issues/1239> as completed.

timoschilling, 2023-04-02:
There is nothing attached

kingwalius, 2023-04-02:
There is a picture attached