mongoose-os-apps/shelly-homekit #1241: NTP regular calls to time.google.com I block external calls

5xPa, 2023-03-07:
I have setup a Firewall rule blocking my IOT devices from the internet.

I then have shelly mongoose devices regularly trying to poll time.google.com
        Mar  7 13:50:46 oor: [FILTER][Block][LAN/RT/VPN->WAN, 26:32:41    ][@S:R=1:2, 192.168.0.181:4097
	Mar  7 13:51:16 oor: Local User (MAC=40-F5-20-00-CD-7E): 192.168.0.181 DNS -> 208.67.222.222 inquire time.google.com
	Mar  7 13:51:16 oor: [FILTER][Block][LAN/RT/VPN->WAN, 26:33:12    ][@S:R=1:2, 192.168.0.181:4097-
	Mar  7 13:51:43 oor: Local User (MAC=40-F5-20-00-CD-7E): 192.168.0.181 DNS -> 208.67.222.222 inquire time.google.com.
I guess it must make the connection to the timeserver if not keep trying

In my router a Draytek. I can forward an external request to an internal address ie
forward time.google.com >>> 192.168.0.37 Synology NAS with a time sever running. The number of polls to time.google.com drops to about 1 per hour.

Mar  7 17:33:25 oor: Local User (MAC=40-F5-20-00-CD-7E): 192.168.0.181 DNS -> 208.67.222.222 inquire time.google.com

It would appear it still passes to time.google.com

Sorry for the longwinded explanation, would it be possible to have an option to manually set the time server?

5xPa, 2023-03-09:
Am I correct that calls are made to NTP servers time.google.com

timoschilling, 2023-03-09:
Yes the Shelly uses time.google.com for time resolving

5xPa, 2023-03-09:
Thanks for the reply. Any chance of making it a manual choice in the setup?

5xPa, 2023-03-14:
Timo another question regarding Time server. 
I have been as mentioned trying to utilise Draytek LAN dns forwarding, to redirect time.google.com to my local time server.
I found some info on a "type 65 Dns request" if I understand this correctly this forces DNS requests to the time sever in the code. 
My question the mongoose firmware issuing a DNS request that forces the request to bypass my LAN and router settings?

Thanks for all the great work and answer my probably fairly basic questions.

timoschilling, 2023-03-14:
I don't know have a deep knowledge about that DNS behaviour, that's functionality of the underlaying [Mongoose OS](https://github.com/cesanta/mongoose-os).

In the next version the setting of a DNS server will be possible. And in the future it will be possible to set the ntp server.

5xPa, 2023-03-14:
Once again thanks for the reply. I was trying to establish if it's Mongoose which was overriding the Lan/Dns forwarding of the router.
I look forward to the next release, I have 21 device a combination of shelly 1, 2.5 and Uni the are very reliable.

nliviu, 2023-09-08:
You can change the ntp server:
```
curl http://shelly_ip/rpc/Config.Set -d '{"config":{"sntp":{"server":"ntp_server"}},"save":true,"reboot":true}'
```

5xPa, 2023-09-08:
nliviu thanks for the info. Unfortunately I am not familiar with "curl" I I have downloaded and installed. Can you confirm this exactly the command I would issue for a shelly device ip 192.168.0.176 setting the NTP server to 192.168.0.37

curl http://192.168.0.176/rpc/Config.Set -d '{"config":{"sntp":{"server":"192.168.0.37"}},"save":true,"reboot":true}'

5xPa, 2023-09-08:
Hi nilivui entered exactly as following
curl http://192.168.0.176/rpc/Config.Set -d '{"config":{"sntp":{"server":"192.168.0.37"}},"save":true,"reboot":true}'

received error message {"code":400,"message":"value is required"}

Googled the error message
The 400 Bad request status code indicates that the server was unable to process the request due to invalid information sent by the client. In other words, the client request needs modification

No idea how to proceed, be grateful for your assistance

nliviu, 2023-09-08:
I suppose you are on Windows:
```
"{\"config\":{\"sntp\":{\"server\":\"time.google.com\"}},\"save\":true,\"reboot\":true}"
```

5xPa, 2023-09-08:
Hi yes I am on windows 11

I entered the following

curl http://192.168.0.176/rpc/Config.Set -d
"{\"config\":{\"sntp\":{\"server\":\"time.myip.com
\"}},\"save\":true,\"reboot\":true}"

No error messages and the router log appears to show the shelly making
time.myip.com requests.

I did try curl http://192.168.0.176/rpc/Config.Set -d
"{\"config\":{\"sntp\":{\"server\":\"192.168.0.37"}},\"save\":true,\"reboot\":true}"

Received error message

On router created time.myip.com 192.168.0.37

Many thanks for your invaluable assistance

JPM-git, 2024-03-26:
I was just searching for something to make use of dns server after 1.5 update but came across this thread with almost 6 months and no replay.

the error is in the command

I did try curl http://192.168.0.176/rpc/Config.Set -d
"{\"config\":{\"sntp\":{\"server\":\"192.168.0.37\"}},\"save\":true,\"reboot\":true}"

you missed a \ after the ip address, " is a special character so you need to "escape" it

if you see close you have it on the fqdn command but when you replace with the ip you wrongly deleted it.