mongoose-os-apps/shelly-homekit #1243: Shelly 1PM reboots after SW relay switch off

mbiskorovajny, 2023-03-10:
I just installed Shelly 1PM, directly flashed it with shelly-homekit and noticed very strange behavior. It reboots when relay is switched off via webUI or Homekit. Physical switch works fine. 

Since it is somehow usable (It is bathroom fan so it is not switched on and off all the time. After reboot in maybe 30 seconds everything works again) I can live with it for some time.

So far I tried Reboot and Reset via webUI. Still the same behavior. I think it is not Homekit connected since the behavior is the same in webUI.

timoschilling, 2023-03-15:
Does the same happen when not load is attached?

mbiskorovajny, 2023-03-16:
In the meantime I found that it could be rc snubber. It is already installed. So far it looks like that was the issue. What I find strange is that physical switch always worked. I will come back if the issue reappears. Thank you

closed 2023-03-16 by timoschilling