mongoose-os-apps/shelly-homekit #1244: Shelly Plus 1PM reboots every 3-15 minutes

rs38, 2023-03-11:
it *was* turning off until I changed the initial state of the shelly from ???  to "on". 
I turned on the file logging with `/rpc/Config.set?reboot=true&save=true&value={"file_logger":{"enable":true}}` it should have captured many of the reboots.
How do I read the file and share the content for debugging?

timoschilling, 2023-03-12:
There is a „Log“ link in the system section of the web Ui. Did you remind why you use a prebuilt and where you downloaded them?

rs38, 2023-03-12:
https://github.com/mongoose-os-apps/shelly-homekit/pull/1197
https://github.com/mongoose-os-apps/shelly-homekit/issues/1194#issuecomment-1420520273

timoschilling, 2023-03-15:
Please post the log of an accident with the current build and then test your Shelly with a regular build.

rs38, 2023-03-16:
does the log via `http://192.168.0.x/debug/log` really persist and continue after a reboot?

timoschilling, 2023-03-16:
Check `/rpc/Config.Get`, if `file_logger.enable` is still `true` it should preserve the log after a reboot. But the file is rotated after some time. So you need to check the log direkt after a reboot.

jeffnhelen, 2023-03-25:
[Shelly 1+PM Persistant Log.pdf](https://github.com/mongoose-os-apps/shelly-homekit/files/11069483/Shelly.1%2BPM.Persistant.Log.pdf)

@timoschilling I posted this in [https://github.com/mongoose-os-apps/shelly-homekit/issues/1238](url) also

I recently updated from 2.11.0-alpha8 as I would periodically (every few days...sometimes longer) have no response issues with HK.  After upgrading to 2.11.2 Master I see even more frequent issues (every few hours...sometimes minutes).  I have enabled the log save option and was able to upload a log file to reference as well.  I would notice that logging would stop refreshing while watching it so it seemed that's when a reboot happened and I would have to start a new log window as soon as I could get back into the main page.

It appears that the last messages show either HomeKit communication or me looking at the webpage (watching the uptime), but then the counter resets and first message is RPC/UART0.  I was able to capture this same event during two crashes:

132053768 mg_rpc.c:314            Shelly.GetInfoExt via WS_in x.x.x.145:64842
139578686 shelly_main.cpp:449     Up 139.57, HAP 0/1/16 ns 1, RAM: 208320/177276; st 40; 0.1: st:0 in_st:0 inm:3 ininv:0; 6.1: svt:6 inv:0 
141384737 mgos_http_server.c:180  0x3ffc7be0 HTTP connection from x.x.x.145:64846
3200073 mgos_rpc_channel_ua:320 0x3ffc79e0 UART0
3262707 shelly_reset.cpp:222    Reboot counter 2
3355345 mgos_sys_config.c:323   Saved to conf9.json
3375599 shelly_main.cpp:796     === Creating peripherals
3384179 shelly_output.cpp:43    OutputPin 1: pin 26, on_value 1, state off
3400856 shelly_input_pin.cpp:44 InputPin 1: pin 4, on_value 1, state off
3406456 shelly_pm_bl0937.cpp:71 BL0937 @ 5/18/23 apc 1.867207
3411753 shelly_noisy_input_:107 Starting sampling timer
3469881 shelly_noisy_input_:118 NoisyInputPin 0: pin 25, on_value 0, state off mc 1 0x2000000
3475303 shelly_main.cpp:255     === Creating accessories
3481875 shelly_switch.cpp:245   Exporting 'Shelly SW': type 0, state: 0
3493188 shelly_main.cpp:304     === Starting HAP bridge (ID XX:XX:XX:XX:XX:XX, UUID <n/a>, CN 11, 2 accessories)
3513500 HAPPlatformTCPStrea:187 Listening on 9000
3523603 mgos_init.c:36          Init done, RAM: 273612 total, 232776 free, 232776 min free
3529797 shelly_main.cpp:341     HAP server state: 1

Not sure if it's unrelated, but most times after reboot the Wifi is disabled and I have to access it via the AP to re-enable Wifi and get it connected back to my network.

kersch79, 2023-04-16:
Similar situation with a shelly1 plus :(

5xPa, 2023-04-16:
Just a thought but do you have a firewall blocking the Shelly from the Internet?As the Shelly will be trying to connect to an NTP serverI blocked mine, and using the router LAN/DNS forwarding to redirect to local NTP serverSent from my iPhone

closed 2023-05-15 by github-actions[bot]

SebastianStolze, 2024-09-11:
Hey I have the same problem with a shelly 1 pm, which was running for 172 days without a problem but now it reset every 1-3 minutes and therefore disabling the light. I first updated to the latest firmware version, without any improvement. Reverting to the stock firmware fixed the issue for now. Is there anything I can do to fix this? 

I also have a second shelly installed which is still working.