mongoose-os-apps/shelly-homekit #1250: Home App slider function broken with Roller shutter(Shelly 2.5) since 2.11.0

macSE1990, 2023-03-24:
I updated 6 Shelly 2.5 from 2.10.1 to the latest release.
Since then when sliding the control in Home App on iPhone to full up or full down position the control "bounces" back. The roller Shutter runs for a second or two, moves a little bit (2%) and stops.
I downgraded the firmware step by step and it seems that this behavior starts with version 2.11.0

davidfierro, 2023-03-28:
I also discovered this behaviour. I'm coming back to 2.10.1 because of that.

marcus-aa, 2023-04-12:
This is most likely the "STOP" behaviour introduced by 

https://github.com/mongoose-os-apps/shelly-homekit/issues/181

It should really be made an option because it breaks the normal way rollers work in Homekit.

closed 2023-05-11 by github-actions[bot]