mongoose-os-apps/shelly-homekit #1254: Shelly25 switches wifi setup ‚off‘ (white) after loss of power and is only available via shelly network only

cl3answ33p, 2023-03-29:
Everything was setup properly. I flashed 6 Shelly 2.5 with homekit and configured them. Afterwards they were integrated into homekit. Shelly‘s have been working for a few weeks but I had to interrupt power on groundfloor. When the power was back 2 Shellys had disappered. One of them I found in the default shelly network (192.168.33.1) by conecting via WLAN by phone and reconfiguring it. I had to ‚enable‘ Wifi1 in Wifi settings section again and save. But again it was not saved, any ideas how to fix this?

jeffnhelen, 2023-03-29:
Have this same issue after flashing +1PM to 2.11.2 and device power cycles on it own.  Oddity is it doesn’t seem to be every power cycle for me.

holypetrus, 2023-04-01:
Hey, i am having the same issues. After power loss one of my Shelly i4s loses its wlan and I have to reconfigure it.

It is very annoying because also in homekit it has to be newly added. All my button shortcuts must be set up again. 

Since we are moving in we have to cut power a few times and every time we need to set it up again. 

I would be glad if there is a solution for this.

danricho, 2023-04-01:
I’ll admit that early on I had this happen on my RGBW2. I haven’t seen it again and I now have a few more RGBW2 and Plus 1.

JanMostert, 2023-04-02:
Exactly the same problem. Never had this problem after a power failure but recently upgraded from 1.8 to latest version (2.11.2) and now nothing but disconnects. Downgrading is impossible because of the continuous disconnects of the Shelly, so sending the URL for the 1.8.1 downgrade never finishes…

cl3answ33p, 2023-04-26:
Hi,
unfortunately there continues to be the same issue with 2 out of 6 shelly25. For one of them it happens from time to time and the first one in this discussion follows the pattern of loosing power and then forgetting the wlan setting up the default shelly wlan.

Cheers
Andreas

cl3answ33p, 2023-05-18:
still the same …

JanMostert, 2023-05-18:
I think it is better to throw them away and buy something else. In my opinion all problems started after I upgraded drom version 1.8. As it is impossible to downgrade, I no longer pit any effort in these devices.Groet,Jan Mostert

jotzet79, 2023-05-29:
Same here, Shelly switch 2.5 - WiFi disconnects every day. There is also the same pattern, where the WiFi setting is not enabled. The issue is not resolved by the second WiFi setting ;(

cl3answ33p, 2023-06-20:
Unfortunately still the same.Von meinem iPhone gesendet

cl3answ33p, 2023-07-12:
still hanging in there …
Waiting for fiX! 🤷🏼‍♂️

cl3answ33p, 2023-08-08:
Unfortunately still the same.

timoschilling, 2023-08-09:
unable to find a solution

closed 2023-08-09 by timoschilling