mongoose-os-apps/shelly-homekit #1256: Shelly Plus Plug S

salee31, 2023-04-02:
I couldn‘t find any comments regarding the Plus Plug S, is there any news regarding this device?
Is there upcoming support for Shelly Plus Plug S device?

tvavrys, 2023-07-15:
Any update on this?

trucidare, 2023-08-03:
Hey, 

created a PR with a basic implementation without LED support. There are i think neopixel stuff used for LED stuff and not figured out how to use them right now.

Currently it support the button, homekit support, power- and temperature measurement.

bsoener, 2023-08-10:
So I can't get my Shelly plug s flashed. Shows no reaction... As soon as I this link: http://192.168.178.66/ota?url=http://rojer.me/files/shelly/shelly-homekit-ShellyPlugS.zip
I get this message:
Trying to update from http://rojer.me/files/shelly/shelly-homekit-ShellyPlugS.zip

When I go back to the Ip of the Shelly no Homekit view comes. Latest original firmware has been updated and wifi connection is strong. Does anyone know what this can be?

raubsau, 2023-08-27:
@bsoener 
Are you trying to flash a Shelly Plug S or Shelly *Plus* Plug S with the Shelly Plug S firmware?

Anyway, try to set a gateway and/or DNS IP in the network section of your device.

bsoener, 2023-08-27:
I'm trying a Shelly Plus Plug S. But from what I've been told, this device is not supported?

bsoener, 2023-09-17:
Is there no solution for this? Or an alternative to flashing?

raubsau, 2023-09-17:
I am waiting for a "native" solution too.
In the meantime, I will use Hombridge (https://homebridge.io/)

bsoener, 2023-09-17:
Ok. I have already installed the plugin on the Homebridge. However, a message appears on the administration page: 

homebridge-shelly
No devices discovered

Can you please tell me what could be the reason for this?

jamesv91, 2023-10-25:
Hi, please is possibility to make flash the Shelly Plus Plug S with Homekit? I have the same issue  like [bsoener](https://github.com/bsoener) commented [on Aug 10](https://github.com/mongoose-os-apps/shelly-homekit/issues/1256#issuecomment-1673796256)

Older Shelly Plug S was OK, also Shelly S - with no problem

raubsau, 2023-10-25:
> Can you please tell me what could be the reason for this?
I cannot help you, sorry...

timoschilling, 2023-11-22:
We currently support only the Shelly Plug S (which is no longer sold by Shelly) and not the new Shelly Plus Plug S!

Ladenburg1, 2023-11-23:
Is there any timewindow when it will be supported or do you have a workaround?
Thanks a lot

gizzy90, 2023-12-26:
> Hey,
> 
> created a PR with a basic implementation without LED support. There are i think neopixel stuff used for LED stuff and not figured out how to use them right now.
> 
> Currently it support the button, homekit support, power- and temperature measurement.

Hey, I really like the work which is done for Shelly Homekit :) And I bought Shelly Plus Plug S and tried to use your PR. But I am not used to Git etc. Tried to "Makefile" a new .Zip to Upload the firmware, but failed. Any chance to get the .zip File from you to flash my shelly?

trucidare, 2023-12-31:
Hey, find the build firmware in the link:

[fw.zip](https://github.com/mongoose-os-apps/shelly-homekit/files/13801381/fw.zip)

build using make ShellyPlusPlugS LOCAL=0

fek89, 2024-01-05:
Hi,
I tried to update the Shelly Plus Plug 1 with this SW from @trucidare but I failed. Could you please explain me how to do this? And does the Plus Plug 1 then work with Homekit?
Thanks a lot!!

theArch86, 2024-01-10:
> Hey, find the build firmware in the link:
> 
> [fw.zip](https://github.com/mongoose-os-apps/shelly-homekit/files/13801381/fw.zip)
> 
> build using make ShellyPlusPlugS LOCAL=0

Sorry, I don't know what I'm doing - but I tried

make ShellyPlusPlugS LOCAL=0

and get 

`make: *** No rule to make target `ShellyPlusPlugS'.  Stop.`

Would you be so kind and tell me what to do to build an firmware for ShellyPlusPlugS

MountainMan1977, 2024-01-21:
Hi, i have the same issue (Plus Plug S v.2 ) - there is a newer Version here in Germany I think only with physical changes - some contacts had not specific length and so on...
Do you think there will be a firmware someday for this plug - otherwise I will send it back?! 
Best regards and thanks for your work you share here!

Ladenburg1, 2024-02-09:
I think there is nothing going on here. Thank you all for your great work using Shellys with Apple Homekit!
I'm switching to original Shelly firmware and integrat it in Homekit using HomeAssistant. This works fine running on a Raspberry.
Best regards

nikeee, 2024-02-29:
I compiled the branch of @trucidare's fork (binary attached here: [shelly-homekit-ShellyPlusPlugS.zip](https://github.com/mongoose-os-apps/shelly-homekit/files/14453341/shelly-homekit-ShellyPlusPlugS.zip))

The RPC websocket returns this error while flashing:
```json
{
    "id": 10,
    "src": "shellyplusplugs-<redacted>",
    "dst": "<redacted>-<redacted>-<redacted>",
    "error": {
        "code": -105,
        "message": "Argument 'key', value 'units' not found!"
    }
}
```

raubsau, 2024-03-01:
> I think there is nothing going on here. Thank you all for your great work using Shellys with Apple Homekit! I'm switching to original Shelly firmware and integrate it in Homekit using HomeAssistant. This works fine running on a Raspberry. Best regards

This is exactly what I did a few weeks ago: Home Assistant (HAOS) in Proxmox, created several Homekit Bridges via configuration.yaml and expose the Shelly Plugs via Home Assistant to Homekit.

As mentioned, thank you to everybody working on a Homekit integration of Shelly devices!

bsoener, 2024-03-01:
Guys, you don't need to ask whether this will be developed any further. As far as I know, the developer is no longer working on this object because he was acquired directly from Shelly and is no longer allowed to do so.

lucassite120, 2024-04-06:
OTA flashing HAA Homekit firmware now is very easy:
https://github.com/RavenSystem/mgos32toHAA?tab=readme-ov-file#shelly-convert-to-haa

markirb, 2024-05-14:
> I compiled the branch of @trucidare's fork (binary attached here: [shelly-homekit-ShellyPlusPlugS.zip](https://github.com/mongoose-os-apps/shelly-homekit/files/14453341/shelly-homekit-ShellyPlusPlugS.zip)) and I'm having trouble flashing it.
> 
> Edit: I seem to have the Shelly Plus Plug S v2 `SNPL-10112EU`: https://kb.shelly.cloud/knowledge-base/shelly-plus-plug-s-v2 This is the model name that is printed on the plug. The device information that one can download in the web interface state that it is a `SNPL-00112EU`. Don't know if this matters, or it means that both devices can use the same firmware.
> 
> After enabling debug logging, I found this message:
> 
> ```
> shelly_ota.cpp:245 Aborting update: backend 0x3ffe0a98 aborts update at file bootloader.bin: -4: app / fs already written, can't update boot
> ```
> 
> I'd like to help with the implementation for v2. Any pointers on how to get information about the hardware? It seems that it's necessary to break the case to take a look at the hardware.
> 
> Edit2: I found this comment regarding the error: [tasmota/mgos32-to-tasmota32#31 (comment)](https://github.com/tasmota/mgos32-to-tasmota32/issues/31#issuecomment-1865993407)
> 
> * It seems that the order of entries in the zip file matters. The manifest must come first and the remaining files must be added in thee same order as they are listed in the manifest file.
> * Additionally, some files must be written before others.
> * I think `boot` and the `partition table` must always come first, as they have to be written before `app` and `fs`. This is also consistent with the manifest of the [official firmware](https://updates.shelly.cloud/update/PlusPlugS).
> * Now the error is `shelly_ota.cpp:245      Aborting update: backend error while processing data: -1 (-10: Cannot relocate shelly)"`, which may be because the new partition table is incompatible with the one of the stock firmware. Or maybe something entirely different.
> 
> Edit3: I managed to flash the firmware by using the fs order from the stock firmware as well as its partition table. By adding another nvs, I seem to have managed to get the same layout as the stock image. However, I won't share the file because I'm not sure how safe it is to use. It may brick devices as soon as something is written to the persistent file system because the addresses are not aligned with the one from the actual build of the firmware.
> 
> So I can confirm that the PR of @trucidare works if the layout is changed. I think this can be fixed by changing the FS config in `mos.yml`, but I don't know how this works. This may be a thing that can be done in 10s if someone knows what they are doing.

Hi,

very good findings. I was just looking into the same things for a ShellyPlus1PM.

I think they changed the filesystem Type:

I had to do this:

```
-        ESP_IDF_EXTRA_PARTITION_2: "shelly,data,nvs,0x3fc000,16K"
+        ESP_IDF_EXTRA_PARTITION_2: "shelly,data,0x88,0x3fc000,16K"
```

So changing the nfs to 0x88. Afterwards it worked.
I will test this and maybe get this firmware running again...

markirb, 2024-05-15:
So  I tested more of this today and this works perfectly for me now.

@trucidare if you want to fix this then just merge https://github.com/mongoose-os-apps/shelly-homekit/pull/1387 into your PR

mjohm1, 2024-05-17:
I don't know how to generate the file myself. It would be so great if somebody could upload the ZIP.

markirb, 2024-05-18:
[shelly-homekit-ShellyPlusPlugS.zip](https://github.com/mongoose-os-apps/shelly-homekit/files/15360899/shelly-homekit-ShellyPlusPlugS.zip)

This is the current build from PR. Try at your own risk

Marcusyo, 2024-05-21:
Is there anything I am doing wrong? I cannot update my plus plug s with the zip file: Aborting update: manifest.json was not first in archive. Is there anything I can do to use the zip file?

markirb, 2024-05-21:
actually it is first in the zip file, strange thing...

[image]

Marcusyo, 2024-05-21:
Maybe its because im on mac and had to create a separate zip file. I will try on my windows computer later

markirb, 2024-05-21:
Yes that’s why. Right click file and say load link then it will not be automatically extracted Von meinem iPhone gesendet

jlanatta, 2024-05-27:
@markirb I tried the build and it works, but I cannot revert to the original firmware (even using the zip file), I noticed you mention this on the [PR](https://github.com/mongoose-os-apps/shelly-homekit/pull/1388). Any hint on what to do?

P.S.: I'm also posting here just for awareness.

markirb, 2024-05-27:
Hi,

good that it works. And bad... but thanks for posting this for awareness.

Yes this issue is also known with 2PM.. so far none of the other Plus models had this issue.
I can only guess that it has to to with the Firmware trying to find some info about the exact device (there are many PlusPlugS versions?!) and we somehow interfere with that information.

 I am still trying to figure it out, but this is a tough one.

jlanatta, 2024-05-27:
**I deleted my previous post (in which I mentioned I had issues reverting to the original firmware) as per I solved it and just noticed this response. So I'm writing again for the sanity of people reading :)**

- I installed the linked firmware and it works fine.
- I was able to revert to shelly's firmware using [this file](https://fwcdn.shelly.cloud/gen2/PlusPlugS/7721b41028e30127f83044c62d410b10b33f948a45046e803d806f8d3fee630c).

So, at least with my model (Plug Plus S v2) everything when OK.

markirb, 2024-06-02:
OK thanks for testing. The downgrade does not work simply because the file is not yet on rojers server.

Apart from that it seems to work for you guys?
Did anyone compare power readings to stock FW?

br

InventoryAndi, 2024-06-04:
For me it does not work well. Flashing worked. It always turns off directly after switching on. So I can't compare power consumption from original firmware. Rebot and initial state on does not help

InventoryAndi, 2024-06-04:
if it helps:
[image]

markirb, 2024-06-04:
That does unforunately not help. Can anyone confirm this? It really looks as though the interface toggles twice (RPC....). Did you try another browser?

InventoryAndi, 2024-06-04:
I tried it via safari and chrome. Same behavior. But I found the issue. I deactivated "State LED" and then this issue happens. When I keep it on, status on / off works.
One more thing to add: when I unplug it I always lose wifi setting. So access is then only possible via direct connection to the shelly afterward. Connect it back to Wifi (when adding all settings) works. But as soon as I unplug it same again (independent if state LED is on or off)

markirb, 2024-06-05:
What do you mean with State LED?

markirb, 2024-06-05:
Ah I see the error. I will fix this

markirb, 2024-06-05:
Test this one:

[release-ShellyPlusPlugS.zip](https://github.com/user-attachments/files/15594508/release-ShellyPlusPlugS.zip)

InventoryAndi, 2024-06-06:
Can't update with this file. I noticed that this one has 822Kb, and the other one which you shared before was 1,6 MB

markirb, 2024-06-06:
This is a zip file containing the zip you have to flash

InventoryAndi, 2024-06-06:
Got it. Updated worked but I have the same issue with the wifi connection. As soon as you unplug it, plug it back, wifi is lost and it is only possible to connect directly to the device and enter back the wifi credentials.

markirb, 2024-06-06:
Sorry the fix was for the status led. For the WiFi there is no difference to other products... so this is not easy. 

Does anyone else here have the same problem?

PR #1305 (2024-06-10): Add basic support for Shelly Plus Plug S
- No LED support

Closes #1256
closed 2024-06-10 by markirb