mongoose-os-apps/shelly-homekit #1258: Help! Shellys won’t get added to HomeKit. Tried everything.

FranciRoosters, 2023-04-06:
Hello,

Please help! I’ve moved to a new home and upgraded my Wi-Fi network to a different mesh and a different SSID. My Shellys, which have already been flashed with HomeKit, won’t be added to the home app despite being toggleable on the web interface. I’ve tried rebooting, resetting and cutting my home power multiple times but to no avail. Additionally, the web interface won’t save my new Wi-Fi network information. What should I do? I’m getting a bit desperate here. Thank you!

bolagnaise, 2023-04-15:
Usually this is related to your wifi settings. HomeKit can be very finicky when it comes to adding devices. I would try experimenting with turning off IGMPV3 (multicast enhancements) and turning off Proxy ARP or similar options (although for me on unifi i have to have proxy arp on in order for the devices to respond correctly). Also make sure you have a seperate SSID for just 2.4Ghz devices to attach to with the channels on 1,6 or 11 and the power set to minimum.

marcus-aa, 2023-04-16:
Mesh creates a lot of problems with multicast, which is essential for Homekit. Mesh should be avoided if at all possible. Turn off all mesh nodes except the one with WAN connection, move the device closer and try again.

marcus-aa, 2023-04-16:
> Usually this is related to your wifi settings. HomeKit can be very finicky when it comes to adding devices. I would try experimenting with turning off IGMPV3 (multicast enhancements) and turning off [...]

Agree. Also "IGMP Snooping" can be something to look for and try different settings.

closed 2023-05-15 by github-actions[bot]