mongoose-os-apps/shelly-homekit #1263: GDO: Wrong initial input state for Shelly1 close sensor

d4rkmen, 2023-04-17:
I have made some researches. The close sensor of Shelly1 has "normally closed" default config. And by default all works as expected. But set it to "normally open" and restart to reproduce the issue. 
After restart the closed sensor state will be: **open** and only return to real state after actual input event (change), so need to open and close gates before it comes normal.
I have made some loggings:

**First opening**
```
3064698144 shelly_main.cpp:443     Up 3064.69, HAP 0/1/12 ns 1, RAM: 28776/7896; st 0; 5.1: cur:open tgt:open cl:0 op:-1
3066529752 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.88.125:62260
3067260923 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.88.125:62260
3068261008 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.88.125:62260
3068516856 mg_rpc.c:310            Shelly.SetState via WS_in 192.168.88.125:62260
3068523143 shelly_output.cpp:63    Output 1: off -> on (GDO:close)
3068530094 shelly_hap_garage_d:307 GDO 1: Tgt State: open -> closed (0 -> 1) (RPC)
3068536802 shelly_hap_garage_d:222 GDO 1: Cur State: open -> closing (0 -> 3)
3068561880 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.88.125:62260
3069029458 shelly_output.cpp:63    Output 1: on -> off (pulse_off)
3069262061 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.88.125:62260
3070267652 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.88.125:62260
3071268294 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.88.125:62260
3072275214 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.88.125:62260
3072698697 shelly_main.cpp:443     Up 3072.69, HAP 0/1/12 ns 1, RAM: 28668/7896; st 0; 5.1: cur:closing tgt:closed cl:0 op:-1
3073272120 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.88.125:62260
3074280141 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.88.125:62260
3075273294 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.88.125:62260
3076270695 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.88.125:62260
3077268619 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.88.125:62260
3078273506 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.88.125:62260
3079270325 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.88.125:62260
3080268893 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.88.125:62260
3080698209 shelly_main.cpp:443     Up 3080.69, HAP 0/1/12 ns 1, RAM: 28668/7896; st 0; 5.1: cur:closing tgt:closed cl:0 op:-1
3081265508 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.88.125:62260
3082266894 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.88.125:62260
3083265315 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.88.125:62260
3084260325 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.88.125:62260
3085259016 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.88.125:62260
3086274343 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.88.125:62260
3087272799 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.88.125:62260
3088270469 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.88.125:62260
3088698400 shelly_main.cpp:443     Up 3088.69, HAP 0/1/12 ns 1, RAM: 28668/7896; st 0; 5.1: cur:closing tgt:closed cl:0 op:-1
3089267414 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.88.125:62260
3090265135 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.88.125:62260
3091261976 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.88.125:62260
3092260051 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.88.125:62260
3093259497 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.88.125:62260
3094274736 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.88.125:62260
3095275541 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.88.125:62260
3096271204 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.88.125:62260
3096698266 shelly_main.cpp:443     Up 3096.69, HAP 0/1/12 ns 1, RAM: 28668/7896; st 0; 5.1: cur:closing tgt:closed cl:0 op:-1
3097269742 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.88.125:62260
3098386354 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.88.125:62260
3099267116 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.88.125:62260
3100264338 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.88.125:62260
3101261594 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.88.125:62260
3102261660 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.88.125:62260
3103261557 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.88.125:62260
3104698309 shelly_main.cpp:443     Up 3104.69, HAP 0/1/12 ns 1, RAM: 28668/7896; st 0; 5.1: cur:closing tgt:closed cl:0 op:-1
3106414922 mgos_http_server.c:180  0x3fff51cc HTTP connection from 192.168.88.125:62973
3106427282 mg_rpc.c:310            Config.get via HTTP 192.168.88.125:62973
3108614676 shelly_hap_garage_d:222 GDO 1: Cur State: closing -> stopped (3 -> 4)
3112698291 shelly_main.cpp:443     Up 3112.69, HAP 0/1/12 ns 1, RAM: 28484/7896; st 0; 5.1: cur:stopped tgt:closed cl:0 op:-1
```

**First closing:**
```
3145261447 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.88.125:62260
3145603903 mg_rpc.c:310            Shelly.SetState via WS_in 192.168.88.125:62260
3145610104 shelly_output.cpp:63    Output 1: off -> on (GDO:open)
3145617246 shelly_hap_garage_d:307 GDO 1: Tgt State: closed -> open (1 -> 0) (fixup)
3145624204 shelly_hap_garage_d:222 GDO 1: Cur State: stopped -> opening (4 -> 2)
3145649253 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.88.125:62260
3146116444 shelly_output.cpp:63    Output 1: on -> off (pulse_off)
3146257492 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.88.125:62260
3147272054 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.88.125:62260
3148269825 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.88.125:62260
3149270310 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.88.125:62260
3150272348 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.88.125:62260
3151271004 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.88.125:62260
3152271281 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.88.125:62260
3152698126 shelly_main.cpp:443     Up 3152.69, HAP 0/1/12 ns 1, RAM: 28768/7896; st 0; 5.1: cur:opening tgt:open cl:0 op:-1
3154343839 mgos_http_server.c:180  0x3fff5134 HTTP connection from 192.168.88.125:62998
3154356136 mg_rpc.c:310            Config.get via HTTP 192.168.88.125:62998
3158271340 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.88.125:62260
3159257626 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.88.125:62260
3160257685 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.88.125:62260
3160698039 shelly_main.cpp:443     Up 3160.69, HAP 0/1/12 ns 1, RAM: 28584/7896; st 0; 5.1: cur:opening tgt:open cl:0 op:-1
3161256527 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.88.125:62260
3168698119 shelly_main.cpp:443     Up 3168.69, HAP 0/1/12 ns 1, RAM: 28768/7896; st 0; 5.1: cur:opening tgt:open cl:0 op:-1
3176698087 shelly_main.cpp:443     Up 3176.69, HAP 0/1/12 ns 1, RAM: 28768/7896; st 0; 5.1: cur:opening tgt:open cl:0 op:-1
3180639043 shelly_input.cpp:76     Input 1: change (state 0)
3180676610 shelly_hap_garage_d:307 GDO 1: Tgt State: open -> closed (0 -> 1) (ext)
3180683745 shelly_hap_garage_d:222 GDO 1: Cur State: opening -> closed (2 -> 1)
```

_Originally posted by @d4rkmen in https://github.com/mongoose-os-apps/shelly-homekit/issues/1024#issuecomment-1511237778_

d4rkmen, 2023-05-09:
Oh I am happy abt it (no I'm not)

timoschilling, 2023-05-09:
reopen the original issue

closed 2023-05-09 by timoschilling