mongoose-os-apps/shelly-homekit #1264: Shelly Devices Initiating Communication on Port 9000

dgvinton, 2023-04-17:
I have a couple of shelly devices on a firewalled "IoT" VLAN.  The Home Hub(s) reside on a "Main" VLAN.  Firewall rules allow all Main->IoT, established Main<->IoT, and IoT->Main on Port 5353.  Everything has been working just fine.

I see, occasionally, Shelly initiated calls to the Home Hub on port 9000 which were dropped.

What is this about?

Thanks,
Don

timoschilling, 2023-04-17:
5353 is just the port for mdns which allows the devices to find each other, the communication itself happens over port 9000. So you need to give it free

closed 2023-04-17 by timoschilling

dgvinton, 2023-04-17:
Hello Timo,

Thank you for the prompt reply and explanation.

I do have one more question.  Why is the Shelly device initiating this
communication?

Best regards,

Don

timoschilling, 2023-04-17:
I don’t know the connection details. We use a reference implementation from Apple called HomeKit ADK. So I expect that it is right.