mongoose-os-apps/shelly-homekit #1273: Feature Request: Shelly 2.5 Only one switch active at a the same time

TheBigNeo, 2023-05-04:
I have a Shelly 2.5 and use it to control my sun blinds. 
Since a second sun blind was subsequently mounted on the same switch, the control is now strange.
When I press open on the switch, the left sun blind opens first and when it is fully extended, the right sun blind opens. 
If I press close, then both close at the same time. 

Therefore, I can not operate Shelly in `Roller Shutter` mode. This leads to the problem that it is possible to activate both switches at the same time, which the two motors do not like at all.

**Feature Request: A setting that even in switch mode only 1 switch can be active at the same time.**

timoschilling, 2023-05-05:
You should solve your problem by using two switches for two shutters.

closed 2023-05-05 by timoschilling

TheBigNeo, 2023-05-05:
I can not use one switch per sun blind. I cannot change this. The wiring in the house is like this. And it's a rental apartment.