mongoose-os-apps/shelly-homekit #1277: [Shelly 1PM] Switch turning off and on by itself

andybenichou, 2023-05-06:
Hi, I'm currently using 10 Shelly 1PM devices in stateless switch mode to trigger automations in Apple HomeKit which control my Philips Hue lights.:
[image]

However, I've been experiencing an issue with nearly all my devices where these modules occasionally turn on and off by themselves. When I manually turn off a module, it often turns back on after a few minutes. Some devices can remain stable for an extended period before suddenly the issue comes back for a few days, after which they return to being stable. Has anyone else experienced a similar issue? The logs are not really informative since it is in stateless mode, I don't see anything printed:

75197616 shelly_debug.cpp:231    No log file, sending new entries
75521177 shelly_main.cpp:449     Up 75.51, HAP 0/1/16 ns 1, RAM: 210108/200540; st 46; 0.1: st:1 in_st:0 inm:3 ininv:0; 3.1: svt:3 inv:0 st:0 m:1 lea: -1.000
83521139 shelly_main.cpp:449     Up 83.51, HAP 0/1/16 ns 1, RAM: 210104/200540; st 46; 0.1: st:1 in_st:0 inm:3 ininv:0; 3.1: svt:3 inv:0 st:0 m:1 lea: -1.000
91521385 shelly_main.cpp:449     Up 91.51, HAP 0/1/16 ns 1, RAM: 210088/200540; st 46; 0.1: st:1 in_st:0 inm:3 ininv:0; 3.1: svt:3 inv:0 st:0 m:1 lea: -1.000
112439237 shelly_main.cpp:449     Up 99.51, HAP 0/1/16 ns 1, RAM: 210108/200540; st 46; 0.1: st:1 in_st:0 inm:3 ininv:0; 3.1: svt:3 inv:0 st:0 m:1 lea: -1.000
112444527 mgos_mongoose.c:66      New heap free LWM: 192204
112453665 mgos_mongoose.c:66      New heap free LWM: 190180
112464261 mgos_mongoose.c:66      New heap free LWM: 188520
119484255 shelly_main.cpp:449     Up 119.47, HAP 0/1/16 ns 1, RAM: 210104/188520; st 46; 0.1: st:1 in_st:0 inm:3 ininv:0; 3.1: svt:3 inv:0 st:0 m:1 lea: -1.000
127484525 shelly_main.cpp:449     Up 127.47, HAP 0/1/16 ns 1, RAM: 210088/188520; st 46; 0.1: st:1 in_st:0 inm:3 ininv:0; 3.1: svt:3 inv:0 st:0 m:1 lea: -1.000

Thanks!

timoschilling, 2023-05-14:
What's the uptime after the shelly automatically turns on? You can see them in the systems section.

andybenichou, 2023-05-14:
> What's the uptime after the shelly automatically turns on? You can see them in the systems section.

Hi, it is really random, it turns on and off after some random time

beroliv, 2023-05-23:
i have the same issue with a shelly 2.5. sometimes i cant switch of the light, and sometimes after switch on the light, shelly switch of by them self.

sebastianheise, 2023-05-24:
The shelly 2.5 issue might be related to what I am seeing: https://github.com/mongoose-os-apps/shelly-homekit/issues/1096

closed 2023-06-22 by github-actions[bot]

marvinjoa, 2024-09-23:
This happens to me as well. Multiple times a week.

markirb, 2024-09-24:
Can you try this: https://github.com/mongoose-os-apps/shelly-homekit/issues/1513

marvinjoa, 2024-09-24:
I just noticed that is not the only issue I have.
It also does not save what is set in Home for "Display as".  It always changes back to switch, when I set it to Light.
I didn't have these issues prior to the latest update.

But I will try to use the power saving setting.