mongoose-os-apps/shelly-homekit #1278: Connect to both HomeKit and Home Assistant to this firmware

WattageGuy, 2023-05-13:
I can't connect to both HomeKit and Home Assistant at the same time, should I be able to do so. Otherwise would it be able to implement such a feature, to be able to connect to both HomeKit and Home Assistant at the same time?

timoschilling, 2023-05-13:
Then someone must write a plug-in for home assistant.

closed 2023-05-13 by timoschilling

gNeeUs, 2023-08-21:
Actually this can be done and it's very easy. Assuming the Shelly device is flashed with the mongoose Open Source Apple HomeKit Firmware for Shelly Devices, first ensure that it is NOT paired with an Apple device via HomeKit. If it is, first Remove the device from the Apple Home app, otherwise you won’t be able to pair it with Home Assistant. For good measure, restart the Shelly device once after removing it. Second, from Home Assistant, the device should have been discovered under Settings > Devices & Services, if it is not, add device Apple Homekit Device and enter HomeKit pairing code XXX XX XXX (obtained from http://ip_of_your_shelly > HomeKit Settings). Finally, ensure to install the Home Assistant integration Apple HomeKit Bridge and configure it. This will in turn pass the Shelly device from Home Assistant to HomeKit on an Apple device such as an iPhone.

timoschilling, 2023-08-21:
With the way @gNeeUs says, the trait of is that the device is only paired with Home Assistant, so you can't use it without a bridge.